OpenXiangShan/XiangShan #1254: `make verilog` print 1 targets failed

eniacL, 2021-11-20:
电脑配置：
虚拟机分配，8核心、40G内存、100存储

Error Log：
Rob: size:256 wbports:18 commitwidth:6
====== Non-inclusive L2 (512KB * 2-bank) prefetch: true ======
bankBits: 1
usr/alias: (2-bit)
usr/needHint: (1-bit)
usr/preferCache: (1-bit)
echo/blockisdirty: (1-bit)
extra ECC Databits:0
extra ECC Tagbits:0
extra ECC Tagbits:0
extra ECC Databits:0
extra ECC Tagbits:0
extra ECC Tagbits:0
	0 <= dcache
	1 <= icache
	2 <= ptw
====== Non-inclusive L3 (2MB * 4-bank) prefetch: false ======
bankBits: 2
usr/preferCache: (1-bit)
echo/blockisdirty: (1-bit)
extra ECC Databits:0
extra ECC Tagbits:0
extra ECC Tagbits:0
extra ECC Databits:0
extra ECC Tagbits:0
extra ECC Tagbits:0
extra ECC Databits:0
extra ECC Tagbits:0
extra ECC Tagbits:0
extra ECC Databits:0
extra ECC Tagbits:0
extra ECC Tagbits:0
	0 <= L2
	1 <= dma ID#0
	2 <= dma ID#1
	3 <= debug
1 targets failed
XiangShan.runMain subprocess failed

Comment：
因为没有具体的报错内容，不知道具体问题是由什么所导致的

poemonsense, 2021-11-20:
可以尝试一下把build.sc里面64G改到32G

eniacL, 2021-11-20:
> 可以尝试一下把build.sc里面64G改到32G

好的已解决

closed 2021-11-20 by eniacL

eniacL, 2021-12-05:
> 可以尝试一下把build.sc里面64G改到32G

# 你好，更新到最新的master之后，还是有相同的报错

# Shell
root@ubuntu:/home/lsf/xs-env/XiangShan# git pull origin master
From github.com:OpenXiangShan/XiangShan
 * branch                master     -> FETCH_HEAD
Already up to date.
root@ubuntu:/home/lsf/xs-env/XiangShan# make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_TRACE=1 -j32
make -C ./difftest emu SIM_TOP=SimTop DESIGN_DIR=/home/lsf/xs-env/NutShell NUM_CORES=1
make[1]: Entering directory '/home/lsf/xs-env/XiangShan/difftest'
make -C /home/lsf/xs-env/NutShell sim-verilog
make[2]: Entering directory '/home/lsf/xs-env/NutShell'
mkdir -p build
mill chiselModule.test.runMain top.TopMain -td build --output-file SimTop.v BOARD=sim CORE=inorder  
[20/85] difftest.resolvedIvyDeps | Downloading [4/4] artifacts (~0/0 bytes)
1 targets failed
difftest.resolvedIvyDeps 
Resolution failed for 1 modules:
--------------------------------------------
  edu.berkeley.cs:chisel3_2.12:3.5-SNAPSHOT 
        not found: /root/.ivy2/local/edu.berkeley.cs/chisel3_2.12/3.5-SNAPSHOT/ivys/ivy.xml
        not found: https://repo1.maven.org/maven2/edu/berkeley/cs/chisel3_2.12/3.5-SNAPSHOT/chisel3_2.12-3.5-SNAPSHOT.pom

make[2]: *** [Makefile:50: build/SimTop.v] Error 1
make[2]: Leaving directory '/home/lsf/xs-env/NutShell'
make[1]: *** [Makefile:29: /home/lsf/xs-env/NutShell/build/SimTop.v] Error 2
make[1]: Leaving directory '/home/lsf/xs-env/XiangShan/difftest'
make: *** [Makefile:111: emu] Error 2

QuramaB, 2023-05-29:
> 可以尝试一下把build.sc里面64G改到32G

你好，我也遇到了一样的问题，将364G修改到32G还是出现了相同的报错，也尝试配置过export _JAVA_OPTIONS='-XX:-UseGCOverheadLimit -Xms256m -Xmx10g -Xss256K'，该问题还是未能解决。
请问还有其它可能的解决办法吗？
错误输出如下：
====== Non-inclusive L3 (1.5MB * 4-bank) prefetch: None ======
bankBits: 2
sets:4096 ways:6 blockBytes:64
[client] size:512.0KB
[client] sets:1024 ways:8 blockBytes:64
blockGranularityBits: 9
usr/preferCache: (1-bit)
echo/blockisdirty: (1-bit)
clientBits: 1
Data ECC bits:8
Tag ECC bits:6
Tag ECC bits:6
clientBits: 1
Data ECC bits:8
Tag ECC bits:6
Tag ECC bits:6
clientBits: 1
Data ECC bits:8
Tag ECC bits:6
Tag ECC bits:6
clientBits: 1
Data ECC bits:8
Tag ECC bits:6
Tag ECC bits:6
        0 <= L2
        1 <= dma ID#0
        2 <= dma ID#1
        3 <= debug
1 targets failed
XiangShan.runMain subprocess failed
make: *** [Makefile:80: build/XSTop.v] Error 1

codersly, 2023-06-18:
你好，我遇到了跟你你一样的问题，请问你解决了吗？ @QuramaB