OpenXiangShan/XiangShan #1257: 请教noninclusive mshr

daxxyy123, 2021-11-22:
在目前a_schedule的流程中，如果acquire请求触发replace，则会对下方产生一个release请求，释放相关的cacheline。然后原先那个cachlineline对应的selfdir、clientdir会被新的cacheline覆盖。但是replace请求按道理是有可能触发向上probetoN操作的，比如本地节点是tip，上方还有一个branch节点，这时候应该把上方branch节点也evict掉。但是现在的流程里并没有这样的处理。而这个流程在inclusive mshr中是有的。
 对于noninclusive mshr，本地节点可以通过clientdir了解上方client的cacheline状态，但是如果跨了一个层级，一个下方节点无法跨层级看到上方节点的情况，例如如下：
clientA 
clientB
clientC
某cacheline在clientB是tip，clientA是branch，当clientB replace该cacheline的时候，如果仅是在本地release，这时候对于clientC来说，它看到的client其实只有clientB，应该看不到clientA，这样它会认为自己的上方client中已经不存在该cacheline了。这样在后面的操作中就可能存在不一致了。
不知道这种情况在noninclusive mshr中是如何考虑的？
谢谢

ljwljwljwljw, 2021-11-23:
> 但是replace请求按道理是有可能触发向上probetoN操作的
- 在non-inclusive cache中，replace可以永远不向上probe，即在replace之后自身失去了一个块，但上方的client中还有该块。保持正确性的关键在于replace时给下层cache的参数，如果参数正确下层cache就可以正确的记录上方的权限。

> 某cacheline在clientB是tip，clientA是branch，当clientB replace该cacheline的时候，如果仅是在本地release，这时候对于clientC来说，它看到的client其实只有clientB，应该看不到clientA，这样它会认为自己的上方client中已经不存在该cacheline了。

- 针对上例，clientB向下replace时，对应参数应该是`max(permA, permB) to permA`， 而不是`max(permA, permB) to N`，通过正确的参数，clientC在接收处理clientB的release后，其记录的client权限实际上是clientA和clientB整体的权限而不是N。

daxxyy123, 2021-11-23:
感谢解释，还有一点没有弄懂：replace之后对应的client dir meta会被新的cacheline 信息覆盖，这样是不是就有可能把原先的client的信息覆盖掉？比如如下场景：
clientA
clientB
clientC
拓扑如上，clientA发出一个acquire请求需要获取cachelineY，该请求在clientB中需要做replace，恰好replace掉的cachlineX在clientA中也有副本，当clientB通过某种方法获得cachelineY后，在clientB中clientA对应的client dir entry中需要写入cachelineY的信息，而这个位置正好又是原来cachelineX记录信息的位置，这样相当于clientA中的cachlineX在clientB中就不可见了。
我没有想明白这种场景是如何处理，似乎在替换client dir entry时，所用的way应该和在clientA中实际存储cachelineY所用的way对应才行，但是这个信息又没有传到clientB。这样好像就可能冲突，能否请帮忙解释一下。
谢谢

ljwljwljwljw, 2021-11-24:
> replace之后对应的client dir meta会被新的cacheline 信息覆盖
client dir依然存在，client dir和self dir是分离的，replace只会invalidate self dir，不invalidate client dir， client dir永远和上层client信息保持一致。
当clientA拿到block Y时，如果blockX仍在clientA中，则clientB中blockX和blockY的client dir entry一定是不同的。

daxxyy123, 2021-11-24:
多谢回复，可能我没有表达清除我的想法，我的意思是clientA从clientB中通过acquire获取了cachelineY，因此这个信息确实应该更新到clientB中的client dir（对应clientA）中，假定是位置U；
但是这个acquire在clientB中导致了replace cachelineX，而这个cachelineX恰好也存在于clientA中，那这个cachelineX在clientB的client dir中自然也是有个位置，假定位置是V，
如果选择的位置U恰好等于位置V，那就相当于clientB在client dir写入cachelineY时，覆盖掉了cachelineX的信息。代码中似乎在选择写入位置上没有做特殊的约束。

daxxyy123, 2021-11-25:
这个问题的本质是如何保证下层client dir中的way和上层self dir中的way一致？如果不一致就会出现问题。

ljwljwljwljw, 2021-11-25:
我理解您的意思了，我们不保证同一个block在client dir和self dir中的way一致，只保证client中的块都在下层cache的client dir里，下层cache分配mshr读dir时，client dir具体选哪个way是根据tag比较结果来的。
例如下面的放分布:

| way | 0 | 1 | 2 | 3 |
|---|---|---|---|---|
| clientA | A | B | C | D |
| clientB(client dir) | D | C | B | A |

daxxyy123, 2021-11-25:
谢谢回复，如果hit那是根据tag比较的结果来选择way，但是如果是miss，那就是replacer选择了，比如考虑这种情况：clientA发出aquire req获取E，在clientB中对于miss请求，下层client dir没有匹配到相关的way后，通过：
val replaceWay = repl.get_replace_way(repl_state)
重新分配使用的way，比如选中了way0，这其实就是要把D replace掉，而在clientA在自己的meta中也需要选一个way，因为是miss请求，而它的way也是通过一个replacer来选择，假如选了1，最后的情况会发现clientA的D在clientB中client dir中没有了，这就有问题了。

way                      | 0  | 1  | 2  | 3
clientA                 | A  | E  | C  | D
clientB(client dir) | E  | C  | B  | A

ljwljwljwljw, 2021-11-25:
> 比如选中了way0，这其实就是要把D replace掉，而在clientA在自己的meta中也需要选一个way，因为是miss请求，而它的way也是通过一个replacer来选择，假如选了1，最后的情况会发现clientA的D在clientB中client dir中没有了，这就有问题了。

当clientB处理acquire blockE时，读client dir，发现读出的client为D(不是invalid项)，这时说明clientB的client dir已经满了，无法继续和clientA保持inclusive关系，此时probeHelper会侦听到这一情况，构造一个probe block D的请求发送到clientA，在probe helper构造的probe做完后(clientA的block D被probe下来)，原处理acquire block E的mshr才会继续开始工作(对应代码`io_probeHelperFinish`)。

目前我们的方案比较保守，即acquire block E等待probe block D完全做完，如果能分析清楚probe block D和acquire block E之间的依赖，可以提高他们的并行度，让这两个请求在一定程度上并行执行(TODO)

daxxyy123, 2021-11-26:
多谢解释，我明白了。

closed 2021-11-26 by daxxyy123