OpenXiangShan/XiangShan #1261: 请教noninclusive mshr中如下代码的问题

daxxyy123, 2021-11-24:
当a_need_data 也就是一个acquire请求，它发起了probe请求，因为存在client dir，这个probe请求应该是很有针对性的，每个收到probe请求的client都是hit的，一定有数据，可是为什么会存在 !resp.hasData的情况？这块一直没有想明白，能否请解释一下。谢谢。
when(req_valid && io.resps.sink_c.valid) {
    val resp = io.resps.sink_c.bits
    probes_done := probes_done | probeack_bit
    w_probeackfirst := w_probeackfirst || probeack_last
    w_probeacklast := w_probeacklast || probeack_last && resp.last
    w_probeack := w_probeack || probeack_last && (resp.last || req.off === 0.U)
    probe_dirty := probe_dirty || resp.hasData
    when (a_need_data && probeack_last && resp.last && !resp.hasData && !nested_c_hit && !self_meta.hit) { //?????
      promoteT_safe := false.B
      s_acquire := false.B
      w_grantfirst := false.B
      w_grantlast := false.B
      w_grant := false.B
      when (!bypassGet && !req_put) {
        s_grantack := false.B
      }
......
      }
    }
  }

linjuanZ, 2021-11-25:
1. 关于 client 返回的 resp 为什么没有 data：client 的权限可能是 B，但是有 cache 别名，需要 L2 把这个块 probe 下来，但是由于上面是 B 权限所以不一定会返回数据的。
2. 为什么只看 probeack_last 决定要不要向下 acquire：这个确实是有问题的。之前的缓存上面只有一个 client，所以没有考虑那么多，现在 ICache 也支持一致性了，这里应该是忘了改了。

daxxyy123, 2021-11-25:
明白了，多谢。

closed 2021-11-25 by daxxyy123