OpenXiangShan/XiangShan #1264: 请教noninclusive mshr中关于will_release_through的问题

daxxyy123, 2021-11-26:
可以产生will_release_through的条件第一个是 !other_clients_hit，第二是虽然 other_clients_hit但是req.param是TtoN或BtoN。我的问题是既然发起req的client申请BtoN，而还至少有一个其它client可以hit，那这个其它client应该也是B，也就是上方除了发起请求的client以外还有至少1个client拥有B权限，那这时候其实这个BtoN的release req似乎也是可以drop，而不必releasethrough?
谢谢。

when(io_releaseThrough && io.dirResult.valid && req.fromC) {
    assert(req_valid)
    // TtoN or BtoN should release through
    will_release_through := !other_clients_hit || (isShrink(req.param) && req.param =/= TtoB) //??
    // report or TtoB will be dropped
    will_drop_release := !will_release_through //
    // if enter this part, we must NOT save release
    // since there is a refill want to replace us
    will_save_release := false.B
    releaseThrough := will_release_through
    releaseDrop := will_drop_release   //会清除sinkC中暂存的release req
  }

linjuanZ, 2021-11-26:
BtoN 且有其他 client 命中的时候确实直接把 Release 丢掉就可以了。

已经在 HuanCun 仓库里更正好了，感谢指出！

daxxyy123, 2021-11-26:
明白了，谢谢。

closed 2021-11-26 by daxxyy123