OpenXiangShan/XiangShan #1265: 请教关于io_is_nestedReleaseData的问题

daxxyy123, 2021-11-26:
io_is_nestedReleaseData是mshrC通知mshr abc，自己是否需要缓存releaseData的信号。如果缓存，mshr abc就需要判断是否存在潜在地址冲突，如果冲突，可能就需要mshrC做releaseThrough。
我的问题是，因为只有releasedata需要真的保存data有地址冲突的情况，但是io_is_nestedReleaseData给出的条件未必一定需要本地缓存Data，比如release parm是BtoN，同时本地hit，这种情况本地本来就有数据，是不需要再保存data的，似乎也就不需要releasethrough？
另外还有一个问题，mshrC会被优先执行，也就是它的releaseData会被先写入，这时候就算和后续req发生冲突，还是有replace流程处理，现在这个releaseThrough的设计是否是处于性能考虑？

io_is_nestedReleaseData := req.fromC && !other_clients_hit /*&& isToN(req.param) */ && req_valid

wakafa1, 2021-11-26:
您说的第一个问题中，“这种情况本地本来就有数据” 的假设可能在嵌套处理的过程中失效。考虑如下情况：
由两个请求发生嵌套，被嵌套者是 acquire block A，嵌套者是 release block C，此时它们对应的本地 meta 都是 block C。在 release 请求看来，此时本地是 hit 的，如果假设此时就不 releaseThrough 了；但是此后被嵌套的 acquire block A 请求可能会写入该数据块（比如 preferCache 或者引发了向上的 probe），在它的视角里，Client 里面还是有 block C 的，因此可以在不做 release 的情况下直接写入，最后当这两个请求全都处理结束时，block C 数据块丢失。

wakafa1, 2021-11-26:
对于第二个问题，我们目前的设计中只会在 dir 返回结果的那一拍就算好是否需要做 replace，因此如果中途突然被其他 req 嵌套是不会有额外的 replace 流程的。为了简化设计避免动态地决定是否要做 replace，同时兼顾性能，我们设计了 releaseThrough 这一套机制。

daxxyy123, 2021-12-01:
谢谢，对于releasethrough这个机制我明白了，但是对于第一个问题还是不甚明了。
对于BtoN，本地hit的情况，因为是B，所以应该是clean数据，此时本地如果有数据也是有效的（本地是B和上一级一样，本地是T就更没有问题了），假如说这个BtoN没有releaseThrough，此时后续的一个acquire因为要用这个单元需要发起replace，单此时这个单元原来的数据会被release到下一级存储单元，并没有发生数据丢失？
我觉得除非release的是脏数据，否则似乎不会发生数据丢失。
能否请帮忙看看。谢谢

wakafa1, 2021-12-02:
在我们现在的 non-inclusive 策略中，如果一个 Acquire block A 本地 miss，假设此时占位这个块是 B，那么它会去检查 B 的 Client 是否存在数据块，如果是的话，Acquire 就**不会**做 Replace，可以直接抹掉本地的 B，因此数据丢失依旧是可能的。
关于脏数据的问题，我们的想法是，即使是干净的数据，不 releaseThrough 确实能保证正确性，但是可能会损失性能，我们希望的是下一级 Cache 有部分充当上一级 Victim Cache 的功效，如果数据丢失则下次访问它需要到更下一级 Cache 中去拿，甚至可能还要额外访问 DDR

wakafa1, 2021-12-02:
有这样几种情况：
1. 本地 B，client B，不会 replace
2. 本地 Tip，一定会 replace
3. 本地 Trunk，不会 replace

daxxyy123, 2021-12-02:
谢谢就最后一种情况，本地B，client B，是不需要release的，但是按照这个条件就是需要做releasethrough了，所以我觉的BtoN似乎应该排除。：
io_is_nestedReleaseData := req.fromC && !other_clients_hit /*&& isToN(req.param) */ && req_valid

wakafa1, 2021-12-02:
我说的 release 指的是 Acquire A 对应的 mshr 的 release，在这种情况下，不需要 release，但是 Release B 对应的 mshr 需要做 releaseThrough

wakafa1, 2021-12-02:
另外还有一点，如果 Acquire A 的 mshr 没有把 B release 下去，然后 Release B 的 mshr 也没有做 releaseThrough，最终的结果是本地和 Client 都没有 B 这个块了，此时看似正确，但下一级的 Cache 的 Dir 维护会出错，因为下一级 Cache 没有收到任何 Release 请求，它还是认为上面的 Client 是包含有 B 这个块的

daxxyy123, 2021-12-02:
多谢，这回明白了，感谢回复。

closed 2021-12-02 by daxxyy123