OpenXiangShan/XiangShan #1271: Where is rob entry allocated and it's structure for csr instructions ?

bhawandeepsingh, 2021-11-28:
Hi,
I am not sure about this info since a lot of people use dispatch, issue etc to mean different things.
I am looking for -
1. Where is the rob entry to an instruction (csr instruction) allocated (in code) ?
2. What is the structure of rob entry in general and csr instruction in particular i.e. what information does it save (in code) ?
Thanks

poemonsense, 2021-11-28:
As you said, people sometimes have different meanings when they refer to dispatch and issue. In XiangShan, we use dispatch for allocating ROB/IQ entries and issue for sending instructions to execution units (alu, load, store, etc).

For question 1), rob entries are allocated in Dispatch [src/main/scala/xiangshan/backend/dispatch/Dispatch.scala](https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/backend/dispatch/Dispatch.scala#L237-L239).

For question 2), the rob entry is instantiated [here](https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/backend/rob/Rob.scala#L335) with the class `RobDispatchData`. We don't have any special data stored in ROB for CSR.

bhawandeepsingh, 2021-12-01:
Thanks @poemonsense 
I also see that CSR instruction must be implemented as fence.
In which stage do you block your CSR instruction so that all preceding instructions have commit? Thanks

poemonsense, 2021-12-01:
Blocked in Dispatch

bhawandeepsingh, 2021-12-05:
@poemonsense Thanks.
Does MicroOp class in Bundle.scala have this information - 
1. is the instruction a csr instruction, 
2. if yes, what is the csr number ?

I am doing a small class project in which I need to stop/ block a csr instruction till a condition is met. All other instructions before the csr instruction may continue to commit in order.
I tried to block it in commit - but apparently rob does not know if it is a csr instruction and what was the csr number
I tried dispatch, but apparently dispatch also does not differentiate a csr instruction from int instruction and hence cannot do it.
I am thinking if this will need to be done in decode, since that has the information but I do not see if decode has ability to block, ( I am trying to figure how it handles if a phy reg or rob is not available, I think that might block in decode). 

Thanks

bhawandeepsingh, 2021-12-05:
@poemonsense Thanks.
Does MicroOp class in Bundle.scala have this information ? 
1. is the instruction a csr instruction,
2. if yes, what is the csr number ?

or how does Dispatch have the info whether its a csr instruction and what is the csr number.

I am doing a small class project in which I need to stop/ block a csr instruction till a condition is met. All other instructions before the csr instruction may continue to commit in order.
I tried to block it in commit - but apparently rob does not know if it is a csr instruction and what was the csr number
I tried dispatch, but apparently dispatch also does not differentiate a csr instruction from int instruction and hence cannot do it.
I am thinking if this will need to be done in decode, since that has the information but I do not see if decode has ability to block, ( I am trying to figure how it handles if a phy reg or rob is not available, I think that might block in decode).

Thanks

closed 2023-02-15 by AugustusWillisWang