OpenXiangShan/XiangShan #1280: 请教一个关于noninclusive mshr中a_schedule流程的问题

daxxyy123, 2021-11-29:
是否要做s_wbselfdir的条件我不太明白，我把它转化一下就是：
!((req.opcode === AcquirePerm && self_meta.miss)|| ((self_meta.miss || req.opcode === Get) && !preferCache))
也就是如果是((req.opcode === AcquirePerm && self_meta.miss)|| ((self_meta.miss || req.opcode === Get) && !preferCache))，就不需要写dir，但是这个好像不大对，比如(req.opcode === AcquirePerm && self_meta.miss)，本地未必不需要写data？？，另外后一个条件只要 miss且!preferCache本地就不应该写cache，为什么还要看是否是get操作？
能否帮忙解释一下，谢谢。
 when (  
      Mux(
        req_acquire,
        Mux(req_needT, !isT(highest_perm_except_me), highest_perm_except_me === INVALID),  
        Mux(req_needT, !isT(highest_perm), highest_perm === INVALID)
      )
    ) {
      s_acquire := false.B
      w_grantfirst := false.B
      w_grantlast := false.B
      w_grant := false.B
      when (!bypassGet && !req_put) {
        s_grantack := false.B
      }
      when(!acquirePermMiss && ((self_meta.hit && !req.opcode === Get) || preferCache)) {//
        s_wbselfdir := false.B
      }
    }

ljwljwljwljw, 2021-11-30:
> 比如(req.opcode === AcquirePerm && self_meta.miss)，本地未必不需要写data?

对于client发下来的acquire perm，client自身只需要perm不需要数据，如果self miss，则可以直接向下acquire perm，最后将perm返回给client，类似于bypass转发，self只起到转发作用。

> 另外后一个条件只要 miss且!preferCache本地就不应该写cache，为什么还要看是否是get操作？

这里是策略问题，和正确性无关，在我们的存储层次中，只有PTW/DMA外设会向下发get，我们不希望这些get污染cache，占用cache的存储空间，因此对于get都不cache，只做数据转发。

daxxyy123, 2021-12-01:
感谢回复。

closed 2021-12-01 by daxxyy123