OpenXiangShan/XiangShan #1281: 请教关于noninclusive mshr中的a_schedule中req_acquire时写selfdir条件问题

daxxyy123, 2021-11-29:
(!acquirePermMiss && (self_meta.hit || preferCache))的条件我不太明白，把这个条件转化一下就是
((req.opcode === AcquireBlock || self_meta.hit) && (self_meta.hit || preferCache)) //因为一定是acquire，所以不是perm就是block
那如果是 AcquireBlock且!preferCache，这种个情况应该不缓存本地，但是按照这个条件判断就是需要写selfdir？？
能否请帮忙解释一下？特别是好几处用了acquirePermMiss，这个的含义不是很明白。
谢谢
when(req_acquire) {
      w_grantack := false.B
      //!acquirePermMiss:req.opcode === AcquireBlock || self_meta.hit
      when(!acquirePermMiss && (self_meta.hit || preferCache)) { //???  如果AcquireBlock且!preferCache如何处理？
        s_wbselfdir := false.B
      }
      s_wbclientsdir := false.B
      when(!clients_meta(iam).hit) {
        s_wbclientstag := false.B 
      }
    }

wakafa1, 2021-11-29:
上面的条件展开应该是  (req.opcode === AcquireBlock || AcquirePerm_Hit) && (self_meta.hit || perferCache) 吧，如果 AcquireBlock 且 !preferCache 则还需要看 self_meta.hit 是否为 true 来决定是否需要设置 s_wbselfdir

wakafa1, 2021-11-29:
如果是 AcquireBlock 且 !preferCache，根据逻辑，需要依 self_meta.hit 来判断是否需要置 s_wbselfdir

daxxyy123, 2021-12-01:
感谢回复

closed 2021-12-01 by daxxyy123