OpenXiangShan/XiangShan #1289: 请教一下huancun中使用的BO prefetcher的问题。

daxxyy123, 2021-12-01:
huncun中使用了BO prefetcher，关于这个prefetcher有2个问题我没有搞清楚，请教一下。
假设L1向下发出hint X地址，默认的bestoffset是2.
这时候记录进入RRtable的值就是X-2；假设真实的offset是3。匹配就是使用offset list中的offset + 收到的地址读RRtable，看是否hit。
如下：
L2依次收到的miss addr：       X+3 |  X+6 |  X+9 ...
BO产生的RRtable匹配addr ： X- 1 |  X     |  X+1 ...
此时虽然offset=3存在于offsetlist中，但是依然没有匹配上。因为每个周期只针对offset list中的一个offset做匹配，即便是L2收到的addr一直是按照表中某个offset产生的，但是只要错过了周期，那就很难再匹配上了。
进一步还有一个问题是假定L1从地址X开始一直按照D偏移产生addr，但是因为收到X时记录的地址是X-D，而而匹配使用addr-offset，那就是说下一次如果可以匹配必须要求收到继续收到X才可以匹配到，但是实际上下一次收到的正确地址其实是X+D，此时即便第一次就用D做匹配，也还是不能命中，这样看似乎会出现永远都不可能再匹配上的情况。
我觉得应该我对算法哪里理解错了，但是看代码确实也是实现的，能否请帮忙解释一下。谢谢。

ljwljwljwljw, 2021-12-01:
您能否再详细的描述一下问题，例如：

> L2依次收到的miss addr： X+3 | X+6 | X+9 ... 
BO产生的RRtable匹配addr ： X- 1 | X | X+1 ...

`BO产生的RRtable匹配addr`指的是为了更新offset list中的每个offset(论文中的`d`)去查RRtable的地址吗？

daxxyy123, 2021-12-01:
BO产生的RRtable匹配addr指的是为了更新offset list中的每个offset(论文中的d)去查RRtable的地址吗？---是的，对应代码中的就是这个输入：
scoreTable.io.req.valid := io.train.valid
  scoreTable.io.req.bits := oldAddr

这个req中的addr在scoreTable中，就是L2收到的miss addr。这个addr减去offsetlist中的某个entry后生成对于RRTable的读地址，根据这个地址是否在RRTable中命中，对于scoretable做记分。

linjuanZ, 2021-12-01:
假设 rrTable 一开始是空的，D 初始值是 2，d 从 1 开始，DCache miss 的步长是 3:
L2 收到的 DCache miss 地址是 X, X+3, X+6, X+9, X+12, ...
每次收到 DCache miss 都要测试一个 d，d 分别是 1, 2, 3, 4, 5, ...
那么每次为了测试 d 访问 rrTable 的地址是 X-1, X+1, X+3, X+5, X+7, ...

假设这些 DCache miss 在 L2 都 miss 了，那么它们各自会有一个预取请求发出去（X+2, X+5, X+8, X+11, X+14, ...），等到这些预取的块拿上来的时候，会把地址 X, X+3, X+6, X+9, X+12, ... 写进 rrTable.

但是 L2 从 L3 把块取上来是需要一定时间的，当 DCache 发来第 3 条请求（X+6）的时候，会测试 X+6-d = X+6-3 = X+3 在不在 rrTable 里面，我们考虑两种情况：

1. L2 已经处理完了第 2 条预取 X+5，并把触发预取的地址 X+3 写进了 rrTable，那么 DCache 发来的第 3 条请求就会在 rrTable 里面命中，d = 3 的 offset 获得 1 分
2. L2 还没有处理完第 2 条预取 X+5，或者已经取上来了但还没有把 X+3 写进 rrTable，那么 DCache 发来的第 3 条请求在 rrTable 里面会 miss，d = 3 的 offset 不得分

第 2 种情况下，虽然 d = 3 没有得分，但是有可能 DCache 发来第 6 条消息（X+15）的时候 L2 做完了第 4 条预取，X+9 在 rrTable 命中，这个时候 d = 6 就会得分，如果还是不够，那么有可能 DCache 发来第 8 条消息（X+21）的时候 L2 做完了第 5 条预取，X+12 在 rrTable 中命中，这时 d = 9 会得分，依次类推。所以这里不存在错过了周期就匹配不上的问题，错过了周期一定是这个 offset 不够用，L2 用 3 作为偏移量来预取跟不上 L1 的速度，所以会在 d = 6, 9, ... 的时候得分，总会有一个合适的 offset，保证 DCache 请求到达 L2 的时候用这个 offset 作为 best-offset 预取一定是足够 timely 的.

linjuanZ, 2021-12-01:
> 此时虽然offset=3存在于offsetlist中，但是因为不可能产生X-2这个地址，所以依然没有hit的结果。因为每个周期只针对offset list中的一个offset做匹配，即便是L2收到的addr一直是按照表中某个offset产生的，但是只要错过了周期，那就很难再匹配上了。
best-offset 里每收到一个 dcache miss 请求测试一个 d，并不是每个周期测试一个 d，所以不会错过每一个 d 的。

> 进一步还有一个问题是假定L1从地址X开始一直按照D偏移产生addr，但是因为收到X时记录的地址是X-D，而匹配使用addr-offset，那就是说下一次如果可以匹配必须要求继续收到X才可以匹配到，但是实际上下一次收到的正确地址其实已经是X+D，此时即便第一次就用offset D做匹配，算得的地址是X，也还是不能命中，这样看似乎会出现永远都不可能再匹配上的情况。
rrTable 里面记录的地址都是预取地址 Y 减去 D，而预取地址 Y 等于 dcache miss 地址 X 加上 D，所以 rrTable 里面大部分情况下其实就是 dcache miss 的地址，只不过是在预取完成的时候才写进来，test d 是用 dcache miss 地址 X 减去 d 之后，跟之前的 dcache miss 地址 X 比较，两者完全有可能是相等的。

daxxyy123, 2021-12-01:
明白了，感谢回复。

daxxyy123, 2021-12-01:
又想到一个问题，对于每一个dcache miss会产生对应的hint，这个机制有个暗示的前提就是每个hint会被及时按序发出，比如X+6会认为X+3的hint一定已经发出，但是目前的hint指令在产生后，被分配个一个mshr，而mshr的调度基本上是一个rrarbiter，先产生的hint未必先发出，实际上发送顺序可能是乱的，这种情况下似乎就会产生偏差？不知道这种偏差是否会影响结果？
谢谢。

linjuanZ, 2021-12-02:
Hint 的发送顺序确实有可能是乱的，dcache 发来的地址也可能是乱的。所以只能看很多轮之后哪一个 offset 的得分最高，从统计意义上讲用多大的步长预取最合适。乱序带来的偏差没有统计过，但是乱序似乎是没法避免的。

另外，香山团队非常感谢您对香山项目的关注，不知道您有加入香山的微信讨论群吗？如果您有意愿共同参与香山开发的话，可以将您的联系方式发送到 [Yinan Xu](https://github.com/poemonsense) 的邮箱。

daxxyy123, 2021-12-03:
多谢解答，这么说在mshr的组织还是用一个环形队列，通过age priority rr arbiter来处理比较合适，我看lsu中其实就是考虑age的，但是到missQueue就是单纯的buffer，不再考虑age了，似乎需要上下一起改，否则prefetch的结果是不可预期的，从占用带宽的角度说可能更差。
我在香山技术讨论二群中，就是现在这个网名，欢迎联系。

closed 2021-12-03 by daxxyy123