OpenXiangShan/XiangShan #1291: 请教noninclusive mshr中w_grant信号的一个问题。

daxxyy123, 2021-12-01:
请问w_grant := req.off === 0.U || io.resps.sink_d.bits.last 中，req.off === 0.U是什么含义？为什么通过这种方式生成w_grant信号？
谢谢
when(req_valid && io.resps.sink_d.valid) {
    when(io.resps.sink_d.bits.opcode === Grant || io.resps.sink_d.bits.opcode === GrantData || io.resps.sink_d.bits.opcode === AccessAckData || io.resps.sink_d.bits.opcode === AccessAck) {
      sink := io.resps.sink_d.bits.sink
      w_grantfirst := true.B
      w_grantlast := w_grantlast || io.resps.sink_d.bits.last
      w_grant := req.off === 0.U || io.resps.sink_d.bits.last  //？？
      bad_grant := io.resps.sink_d.bits.denied
      gotT := io.resps.sink_d.bits.param === toT
      gotDirty := io.resps.sink_d.bits.dirty
    }
    when(io.resps.sink_d.bits.opcode === ReleaseAck) {
      w_releaseack := true.B
    }
  }

linjuanZ, 2021-12-02:
这里参考了 block-inclusive-sifive 的写法。w_grant 表示收到了 Grant，并且可以向上层返回 Grant 了。当 A 通道请求的地址是 block 对齐的时候，可以允许虫洞路由，L2 收到 L3 第一个 beat 的数据就可以返回给 L1 了，不需要等所有 beat 的数据都拿到再返回 L1。但其实香山这里也不支持非对齐的地址，所以这个条件其实是没用的。

daxxyy123, 2021-12-03:
明白了，多谢。

closed 2021-12-03 by daxxyy123