OpenXiangShan/XiangShan #1292: 请教关于noninclusive mshr中  io.status.bits.nestB 的问题。

daxxyy123, 2021-12-01:
io.status.bits.nestB这个信号是abc mshr告知冲突的bc mshr，它已经完成所有b通道操作，bc_mshr可以安全执行，可以是为什么要加入 (!w_grantfirst || (client_dir_conflict && !probe_helper_finish))这个条件？ (w_releaseack && w_probeacklast)不是已经表明它完成所有b通道操作了么？能否请帮忙解释一下，谢谢。

  io.status.bits.nestB := meta_valid &&          //for abc mshr告知潜在冲突的bc mshr，当前状态bc mshr是否可以安全执行。
    (w_releaseack && w_probeacklast) &&          //虽然有b通道操作，但是已经完成
    (!w_grantfirst || (client_dir_conflict && !probe_helper_finish))

ljwljwljwljw, 2021-12-02:
`!w_grantfirst` -> 假设不加这个条件，且当前acquire(abc mshr)和probe(bc mshr)的是同一个块，`w_grantfirst`为`true`，`w_grantlast`为false，即acquire收到了第一个beat，但没有完全收完，那么，如果让b打断a，probe ack data时去读sram，读出的就可能是一个错误的数据，该数据的一部分是grant上来的新值，另一部分是原来sram中的老值，这样就破坏了cache一致性。

`client_dir_conflict && !probe_helper_finish` -> 由于我们的client dir conflict的处理比较保守，在client dir冲突时要等probe helper构造的probe做完，才开始原来的请求，而在这种处理方式下，w_grantfirst可能一开始不会被置，`!w_grantfirst`条件一直是`false`，导致probe helper构造的b请求一直无法嵌套当前请求，从而两者互相依赖，形成死锁。为了破解这一问题，额外加入了`client_dir_conflict && !probe_helper_finish`条件来打破死锁。

daxxyy123, 2021-12-03:
多谢解释。

closed 2021-12-03 by daxxyy123