OpenXiangShan/XiangShan #1310: not found: /root/.ivy2/local/edu.berkeley.cs/chisel3_2.12/3.5-SNAPSHOT/ivys/ivy.xml

eniacL, 2021-12-05:
# CMD：
`make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_TRACE=1 -j32`

# ERROR log:
```
1 targets failed
difftest.resolvedIvyDeps 
Resolution failed for 1 modules:
--------------------------------------------
  edu.berkeley.cs:chisel3_2.12:3.5-SNAPSHOT 
        not found: /root/.ivy2/local/edu.berkeley.cs/chisel3_2.12/3.5-SNAPSHOT/ivys/ivy.xml
        not found: https://repo1.maven.org/maven2/edu/berkeley/cs/chisel3_2.12/3.5-SNAPSHOT/chisel3_2.12-3.5-SNAPSHOT.pom

make[2]: *** [Makefile:50: build/SimTop.v] Error 1
make[2]: Leaving directory '/home/lsf/xs-env/NutShell'
make[1]: *** [Makefile:29: /home/lsf/xs-env/NutShell/build/SimTop.v] Error 2
make[1]: Leaving directory '/home/lsf/xs-env/XiangShan/difftest'
make: *** [Makefile:102: emu] Error 2
```

# Comment:
想请教下，在生成仿真程序的时候。产生了上述not found的报错。是需要对配置进行修改还是其他的处理方式。还是什么东西没有安装？

poemonsense, 2021-12-05:
请尝试把difftest和XiangShan都更新至最新的master分支

eniacL, 2021-12-05:
> 请尝试把difftest和XiangShan都更新至最新的master分支

# 你好，更新到最新的master之后，还是有相同的报错

# Shell
```
root@ubuntu:/home/lsf/xs-env/XiangShan# git pull origin master
From github.com:OpenXiangShan/XiangShan

branch master -> FETCH_HEAD
Already up to date.
root@ubuntu:/home/lsf/xs-env/XiangShan# make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_TRACE=1 -j32
make -C ./difftest emu SIM_TOP=SimTop DESIGN_DIR=/home/lsf/xs-env/NutShell NUM_CORES=1
make[1]: Entering directory '/home/lsf/xs-env/XiangShan/difftest'
make -C /home/lsf/xs-env/NutShell sim-verilog
make[2]: Entering directory '/home/lsf/xs-env/NutShell'
mkdir -p build
mill chiselModule.test.runMain top.TopMain -td build --output-file SimTop.v BOARD=sim CORE=inorder
[20/85] difftest.resolvedIvyDeps | Downloading [4/4] artifacts (~0/0 bytes)
1 targets failed
difftest.resolvedIvyDeps
Resolution failed for 1 modules:
edu.berkeley.cs:chisel3_2.12:3.5-SNAPSHOT
not found: /root/.ivy2/local/edu.berkeley.cs/chisel3_2.12/3.5-SNAPSHOT/ivys/ivy.xml
not found: https://repo1.maven.org/maven2/edu/berkeley/cs/chisel3_2.12/3.5-SNAPSHOT/chisel3_2.12-3.5-SNAPSHOT.pom

make[2]: *** [Makefile:50: build/SimTop.v] Error 1
make[2]: Leaving directory '/home/lsf/xs-env/NutShell'
make[1]: *** [Makefile:29: /home/lsf/xs-env/NutShell/build/SimTop.v] Error 2
make[1]: Leaving directory '/home/lsf/xs-env/XiangShan/difftest'
make: *** [Makefile:111: emu] Error 2
```

poemonsense, 2021-12-05:
从log上看，应该是NOOP_HOME的环境变量设置有问题。需要参考香山README设置变量
Set environment variable NEMU_HOME to the absolute path of the NEMU project.
Set environment variable NOOP_HOME to the absolute path of the XiangShan project.

eniacL, 2021-12-05:
> 从log上看，应该是NOOP_HOME的环境变量设置有问题。需要参考香山README设置变量 Set environment variable NEMU_HOME to the absolute path of the NEMU project. Set environment variable NOOP_HOME to the absolute path of the XiangShan project.

- 已check到式NOOP_HOME错误指向Nutshell，修正即可正常运行。Isuue close

closed 2021-12-05 by eniacL