OpenXiangShan/XiangShan #1328: 请教一个关于香山sbuffer的问题

daxxyy123, 2021-12-09:
我看香山的说明现在已经支持双核了，可是香山的sbuffer和L1 cache之间并没有snoop通道。考虑以下场景来自core2的针对地址A的snoop请求经过共享L2转发来到core1的L1 cache，但是core1针对地址A的写操作还存在sbuffer中还没有进入core1的L1，那这个数据其实就对于core2不可见了？这样似乎就有问题了。不知道是否存在这个问题？谢谢。

wakafa1, 2021-12-09:
是的，因为 RISC-V 实现弱一致性，只有显式地用 fence 指令才会将 sbuffer 中的内容刷到 dcache 中

wakafa1, 2021-12-09:
RISC-V 实现弱一致性，在您说的场景下，需要软件显式执行 fence 指令

daxxyy123, 2021-12-09:
从效率的角度说，fence操作也只能针对特定共享数据去做，但是按照目前的方式双核之间的几乎所有的共享数据操作都普遍的需要使用显式fence指令做同步，这样双核工作的性能是不是会受到一定的影响？另外个人理解relaxed model似乎是在memory consistency方面放松了访存操作之间的次序，但是对于cache coherency这并不是它需要规定的内容（不过我觉得它还是需要建立在cache一致性保证的前提下，如果每一次的写操作都无法保证全系统可见，其实再讨论访存操作的次序问题是比较尴尬的，因为从系统角度说某些写操作会暂时性的消失了），这是我个人的理解，可能不对，请指教。

wakafa1, 2021-12-09:
1. 在我的理解里，弱一致性模型中，一个写操作无法保证全系统可见是很正常的事情，多核编程需要摈弃的一个观念就是可见性；甚至在 TSO 模型中可见性仍然会消失；一个经典的例子是核 0 运行 x := 1; print(y); 核 1 运行 y := 1; print(x); 如果一个写操作一定全系统可见，就不会发生打印 x = 0, y = 0 的情况，但在 x86 机器中运行它都是有概率发生的
2. Cache Coherence 和这个问题关系不大，主要定义在 Cache 层面的一致性
2. 效率问题是存在的，但我个人觉得可能重点在于目前实现所有 fence 操作的方法都是刷掉所有的 sbuffer，可能粒度过大

wakafa1, 2021-12-09:
指教不敢
1. 在我的理解里，弱一致性模型中，一个写操作无法保证全系统可见是很正常的事情，多核编程需要摈弃的一个观念就是可见性；甚至在 TSO 模型中可见性仍然会消失；一个经典的例子是核 0 运行 x := 1; print(y); 核 1 运行 y := 1; print(x); 如果一个写操作一定全系统可见，就不会发生打印 x = 0, y = 0 的情况，但在 x86 机器中运行它都是有概率发生的
2. Cache Coherence 和这个问题关系不大，主要定义在 Cache 层面的一致性
2. 效率问题是存在的，但我个人觉得可能重点在于目前实现所有 fence 操作的方法都是刷掉所有的 sbuffer，可能粒度过大

daxxyy123, 2021-12-14:
多谢回复。
这个例子我没有想太清楚，我觉得可能会有其它的解释，因为x86是tso的，store-load顺序可以颠倒，所以print（y）可能会在x：=1，之前执行；同理print（x）也有可能在y：=1之前执行，所以出现x=0和y=0的情况，可能是因为写操作在做读操作之前还没有操作，未必说明x86的sbuffer全局不可见吧。
另外正如您说的“一般意义来说处理器最终面对的是程序，Memory Consistency 可以作为程序员写多核程序的一种手册是必须要遵守的，而 Cache Coherence 则是为了更方便实现Memory Consistency 而在 Cache 层面运用的一种手段，它对程序是透明的”，我赞成这个观点，所以我觉得risc-v的weak memory model没有一定硬性要求sbuffer不能全局可见，这只是微架构设计的一种选择。同时正如您说“所有 fence 操作的方法都是刷掉所有的 sbuffer，可能粒度过大”，用fence不光是粒度大，对于性能的打击也非常大，利用fence做同步处理都是迫不得已的选择，需要付出较大的性能代价，需要尽可能避免，如果可以通过sbuffer可见的方法，解决一部分核间数据同步的问题，这当然是一个更好的选择。比如dpdk pipeline模式时，一核读一核写是个常见场景，此时共享队列如果需要不断地使用fence指令，系统表现可能就太不理想了，这也就是我认为的store buffer支持snoop会对于多核应用更有意义的原因。
谢谢

wakafa1, 2021-12-15:
我个人觉得，在一核读一核写这样的并发场景中，软件应该尽可能地对一些关键控制变量使用原子读写操作或者采用无锁数据结构，这可以避免大量的 fence 操作；如果增加 Store Buffer 的 snoop 会在硬件中会带来额外的开销（延迟+面积），这可能需要做一个权衡

daxxyy123, 2021-12-16:
是这样的，但是如果storebuffer不可见，这些操作就很难了，因为永远不知道一个已经返回的写操作何时对系统其它部分可见，假如变量A已经被核1改变，改变的值存在storebuffer中，而此时core2发起的snoop只能看到core1的L1 cache中的变量A的原值，读取的也只能是原值，而core2还不知道，这样从正确性考虑可能都会有问题吧，所以唯一可靠的使用共享数据的方法就是用低效的fence操作。个人观点从广义上说storebuffer就是cache的一部分，storebuffer不支持snoop和cache不支持snoop最终的效果其实是一样的。对单核应用storebuffer当然不需要snoop，没有必要付出代价；但是对多核应用这可能就是刚需了。

wakafa1, 2021-12-16:
这里我持保留意见，fence并不是唯一可靠的使用共享数据的方法，因为原子指令是不会走 SBuffer 的，这适合小粒度数据的频繁竞争读写；对于大量数据的共享，比如一个很大的共享数据队列，首先可以用原子读写更新 Index，然后每个核自己操作属于自己范畴的数据，本身操作这些数据就会占用比较长的时间，额外增加一个 fence 并不会有很大的开销

wakafa1, 2021-12-16:
> 多谢回复。 这个例子我没有想太清楚，我觉得可能会有其它的解释，因为x86是tso的，store-load顺序可以颠倒，所以print（y）可能会在x：=1，之前执行；同理print（x）也有可能在y：=1之前执行，所以出现x=0和y=0的情况，可能是因为写操作在做读操作之前还没有操作，未必说明x86的sbuffer全局不可见吧。 另外正如您说的“一般意义来说处理器最终面对的是程序，Memory [...]

另外在这个例子里，无论理由是 Sbuffer 的可见性还是 load-store 的顺序颠倒，对于程序员来说，x=0 和 y=0 的情况是我不愿意看到的，因为它违背了**写可见性**，因此我仍然不得不增加 fence 指令来确保可见性

qianlong-zhang, 2022-03-09:
> 所以print（y）可能会在x：=1，之前执行；同理print（x）也有可能在y：=1之前执行，所以出现x=0和y=0的情况，可能是因为写操作在做读操作之前还没有操作，未必说明x86的sbuffer全局不可见吧。

我认为这里的理解是有问题的，这个例子的前提是TSO读写之间不会乱序，而且是**对程序执行结束后的结果做判断，并不是执行的中间结果**，按照你的理解，“_写操作在读操作之前还没有完成，也就是先执行读后执行写_”，这种是store/load乱序，不满足TSO模型。而即使TSO模型情况下，也会出现x=0;y=0只有一种可能，**即因为store buffer的存在。**

把执行编号方便解释（S1）x=1；（L1） print(y); (S2) y=1; (L2) print(x)；其中可能性如下：
1. S1 L1  S2 L2：结果：x=1，y=0;  x=0, y=0
2. S1 S2 L1 L2：结果：x=1，y=1；x=0, y=0
3. S2 L2 S1 L1：结果：x=0，y=1;  x=0; y=0
结果有可能有这两种的原因是，前者从store buffer出去到L1 cache了，而后者全零是因为还没有从store buffer出去，并且store buffer是单核private的。

另外，对于_"但是按照目前的方式双核之间的几乎所有的共享数据操作都普遍的需要使用显式fence指令做同步"_ 双核之间的共享数据，本来就应该程序员通过显示的指令保证同步，这样并不会对性能有多大的影响，原因在于：
1. 多核之间的共享数据读写都是先争抢写锁后再真正写，而不会出现对于共享数据去，不取枪锁直接开始写，这样大概率会出问题。因此性能是否变差主要在于对于锁的操作是否频繁
2. 问题转换成，如何降低锁操作的代价，有几种办法：原子指令实现所谓的无锁编程；调整共享数据区，尽量减少抢锁操作；区分读写锁；其他手段在某些科研论文中也有提及；

closed 2023-05-22 by wakafa1

Jeremy-Jia, 2024-06-11:
nice question

rzbdz, 2024-11-26:
> 3\. 目前实现所有 fence 操作的方法都是刷掉所有的 sbuffer

你好，请教一个问题，fence 似乎只规定了LD ST的顺序限制，是否实现时，fence执行后时刻并不需要保证fence前的LD/ST的全局可见性落实，而原子指令需要保障全局可见性。

Ma-YX, 2024-11-26:
> 你好，请教一个问题，fence 似乎只规定了LD ST的顺序限制，是否实现时，fence执行后时刻并不需要保证fence前的LD/ST的全局可见性落实，而原子指令需要保障全局可见性。
是的，fence 要求特定访存操作有强制顺序约束，但对全局可见性没有强制要求；原子指令需要保证访存操作的全局可见性