OpenXiangShan/XiangShan #1333: 请教关于dcache中ld、sc指令处理的问题。

daxxyy123, 2021-12-09:
dcache中ld/sc指令的相关代码如下：
 val lrsc_count = RegInit(0.U(log2Ceil(lrscCycles).W))
  val lrsc_valid = lrsc_count > lrscBackoff.U
  val lrsc_addr  = Reg(UInt())
  val s3_lr = !s3_req.probe && s3_req.isAMO && s3_req.cmd === M_XLR
  val s3_sc = !s3_req.probe && s3_req.isAMO && s3_req.cmd === M_XSC
  val s3_lrsc_addr_match = lrsc_valid && lrsc_addr === get_block_addr(s3_req.addr)
  val s3_sc_fail = s3_sc && !s3_lrsc_addr_match
  val s3_sc_resp = Mux(s3_sc_fail, 1.U, 0.U)

  val s3_can_do_amo = (s3_req.miss && !s3_req.probe && s3_req.source === AMO_SOURCE.U) || s3_amo_hit
  val s3_can_do_amo_write = s3_can_do_amo && isWrite(s3_req.cmd) && !s3_sc_fail

  when (s3_valid && (s3_lr || s3_sc)) {
    when (s3_can_do_amo && s3_lr) {
      lrsc_count := (lrscCycles - 1).U
      lrsc_addr := get_block_addr(s3_req.addr)
    } .otherwise {
      lrsc_count := 0.U
    }
  } .elsewhen (lrsc_count > 0.U) {
    lrsc_count := lrsc_count - 1.U
  }
第一个问题是lrscCycles为啥是100，lrscBackoff为啥是3，当然如注释所说指令集定义是ld、sc之间最多16 instructions，但是我不太明白为什么，也不太明白怎么就由此算出了lrscCycles和lrscBackoff。
第二个问题是请问在一条ld指令之后是否会再出现一条ld指令，出现嵌套？

谢谢

AugustusWillisWang, 2021-12-09:
第一个问题:

lrscCycles 是为了支持 RISC-V spec 定义的 `constrained LR/SC loops` (参见 Volume I: RISC-V Unprivileged ISA: 8.3 Eventual Success of Store-Conditional Instructions). 这里的 100 是设计时任意指定的, 为了确保在有部分分支预测错误的情况下16条指令的循环必定能执行完.

关于 lrscBackoff, 这是历史遗留的冗余代码. rocket 中使用这一机制来保证在出现大量 LR 时,  probe 能有机会执行. 香山目前没有这一机制.

第二个问题: 关于 LR / SC, 手册给出了这样的要求:
```
An SC can only pair with the **most recent LR** in program order. 
```

AugustusWillisWang, 2021-12-13:
第一个问题： 16 条指令这个数字来自于手册中的要求，如下图所示。
[image]
确切地说，100 这个数字保证的是在 100 周期内，上图中定义的 `constrained LR/SC loop` （最多 16 条指令）一定能执行完。在这段时间内，对 LR 所设置的 reservation set 中地址的 probe 操作将会被阻塞。当前 hart 能不被打扰的执行一个成功的 SC。

第二个问题：由于手册给出了这样的要求: `An SC can only pair with the **most recent LR** in program order`，`An SC may succeed only if ... and if there is no other SC between the LR and itself in program order.`，所以 lr lr sc sc 这样的指令序列，实际有可能成功的只有中间的这一对指令 lr **lr sc** sc。目前，第二个 LR 出现之后， LR 指令所分配的 reservation set 会被更新。同时， SC 指令不论成功与否，都会清除现有的 reservation set。

AugustusWillisWang, 2021-12-24:
#1383 添加了 lrsc backoff 机制。现在香山可以在双核场景下存在大量连续的 lr 时回避活锁。

closed 2023-02-15 by AugustusWillisWang