OpenXiangShan/XiangShan #1385: Make EMU Error

pengshihan10, 2021-12-22:
我在运行完指令“make emu CONFIG=MinimalConfig EMU_THREADS=2 -j8”后遇到如下报错：

clang++    axi4.o common.o compress.o device.o flash.o keyboard.o ram.o sdcard.o uart.o vga.o difftest.o goldenmem.o interface.o nemuproxy.o ref.o emu.o main.o snapshot.o verilated.o verilated_dpi.o verilated_threads.o VSimTop__ALL.a   -lpthread -lSDL2 -ldl -lz  -pthread -lpthread -latomic   -o /data/ns-fpga/XiangShan/build/emu
rm VSimTop__ALL.verilator_deplist.tmp
make[2]: Leaving directory '/data/ns-fpga/XiangShan/build/emu-compile'
make[1]: Leaving directory '/data/ns-fpga/XiangShan'
ls build
emu				   lock-emu	     SimTop.fir       time.log
emu-compile			   plusarg_reader.v  SimTop.v
firrtl_black_box_resource_files.f  SDHelper.v	     SimTop.v.conf
FlashHelper.v			   SimTop.anno.json  SimTop.v.sram.v
./build/emu -i temp --diff=/data/ns-fpga/NEMU/build/riscv64-nemu-interpreter-so -s 141 -b 0 -e -1   
Emu compiled at Dec 22 2021, 16:33:31
Using seed = 141
The image is temp
Using simulated 8192MB RAM
Can not open 'temp'
emu: /data/ns-fpga/XiangShan/src/test/csrc/common/ram.cpp:138: void init_ram(const char *): Assertion `0' failed.
make: *** [verilator.mk:126: emu] Aborted (core dumped)

请问一下出现这个报错的原因是什么呢？

poemonsense, 2021-12-22:
从命令里面看，`./build/emu -i temp --diff=/data/ns-fpga/NEMU/build/riscv64-nemu-interpreter-so -s 141 -b 0 -e -1`

pengshihan10, 2021-12-22:
在生成的build文件夹中确实没有生成temp文件，这个temp是什么时候并且以什么方式出现的呢？

poemonsense, 2021-12-22:
我们提供了预编译好的bin文件，在ready-to-run目录下面，也可以参考[这个文件](https://github.com/OpenXiangShan/XiangShan-doc/blob/main/tutorial/%E9%A6%99%E5%B1%B1%E9%83%A8%E7%BD%B2%E6%95%99%E7%A8%8B.pdf)的3.1节（使用AM生成workload）利用nexus-am编译一些简单的测试用例。bin文件存储了从0x8000_0000地址开始的数据。

pengshihan10, 2021-12-22:
我在Scala源码中重新配置了香山处理器的地址空间，也使用riscv-pk生成了新的bbl.bin文件，那么我现在要进行仿真下一步需要做什么呢？

poemonsense, 2021-12-22:
我们的框架里面，-i输入的bin文件会被放到0x8000_0000开始的位置。如果修改了起始地址，应该是需要同步修改一下我们的配置，包括仿真SimTop顶层中有一个地址的路由（XSTop那一层会指定几个axi的地址区间，然后外面仿真wrapper SimTop例化RAM的时候也会指定地址区间）。需要仿真可能需要同时改动好几个地方

pengshihan10, 2021-12-22:
问题1：同步修改配置是对src/test文件夹下面的Scala文件中的某个部分进行修改吗？改动的具体地方可以指明一下在哪里吗？
问题2：同步修改配置完成后，是使用Verilator是生成香山核的仿真程序吗？
问题3：如果问题2正确，那是否输入指令“./build/emu -i bbl.bin --diff=riscv64-nemu-interpreter-so -s 141 -b 0 -e -1”就可以得到仿真结果了呢？
问题4：是否可以不接入香山差分仿真测试中，去掉“--diff=riscv64-nemu-interpreter-so ”这个部分？

poemonsense, 2021-12-22:
关于这几个问题，
问题1：我们尽快确定一下这个问题，看看涉及多大改动，scala部分的改动应该不会太大
问题2、3、4：是的

我目前想到的方案是，在src/test下面SimTop中，把AXIRAM的起始地址改到4000_0000，同时修改src/main下面PMA相关的配置。然后在difftest/src/test中，把RAM的起始地址同步改到4000_0000，这样应该就能仿真了，在`--no-diff`选项下。具体我需要一些时间来确认一下，您可以先试试看

pengshihan10, 2021-12-23:
那就麻烦同学尽快确定下了，我这边也按照你说的方式尝试一下。这边因为我使用的是“5c7674fe”版本，可能和现在的文件分布有区别。

src/main文件夹下面的改动，应该是属于香山核本身的改动，我这边已经完成了，应该就是包含PMA.scala的MEMMAP更新以及Top.scala中对地址空间的重新分配以及修改ResetVector的数值。

主要还是仿真环境这边的改动，不知道我对你的意思理解有误没有，其实就是把所有原先设置为8000_0000都更改为4000_0000。还有在src/test也包含了大量的.cpp和.h文件，这里面的参数需要修改吗？

pengshihan10, 2021-12-23:
这个是我截取的关于8000_0000的所有相关内容，只要把它们都替换成4000_0000，是否就可以进行仿真了呢？
[image]

pengshihan10, 2021-12-23:
还有个很奇怪的现象：我生成的仿真程序并没有--no-diff这条指令
[image]

Lemover, 2021-12-23:
你好，我修改了一个简单的版本，可以从0x40000000的内存取指并正常执行，在reset-vector分支，可以用作参考。（修改的比较灵车，不适合照搬）

pengshihan10, 2021-12-23:
> 你好，我修改了一个简单的版本，可以从0x40000000的内存取指并正常执行，在reset-vector分支，可以用作参考。（修改的比较灵车，不适合照搬）

好的，谢谢同学，我参考一下

pengshihan10, 2021-12-23:
我之前按照poemonsense说的方法自己修改后跑了一下仿真，但是执行后一直如下的信息在刷屏，几乎没有可读性，这样是证明正常跑起来了吗还是说要执行到某个操作停止后生成某些文件呢？

我的执行指令为“./build/emu -i bbl.bin”

[image]
[image]
[image]

Lemover, 2021-12-23:
这是程序执行完之后的性能计数器结果，“大概率”执行成功了。执行时可以添加参数 -e 0 避免性能计数器的打印。

pengshihan10, 2021-12-23:
添加参数 -e 0后，只有“[warning] sdcard img not found”打印。

强制结束后，会有如下信息打印，请问一下这个结果是什么意思呢？

[image]

pengshihan10, 2021-12-23:
另外，经过对比Lemover同学释出的版本，我发现我们之间的唯一的区别是“assert(bus.a.bits.address >= 0x80000000L.U)”、“assert(bus.b.bits.address >= 0x80000000L.U)”、“assert(bus.c.bits.address >= 0x80000000L.U)”我没有修改，这个部分具体作用是什么呢，它会实际影响到仿真以及实际上板调试的哪些数据呢？

poemonsense, 2021-12-23:
现在应该是能正常仿真的。可以通过rob/roq的io*pc信号，结合io*commits*valid和walk信号，来确认一下，指令序列是否符合预期

pengshihan10, 2021-12-23:
仿真的话应该怎么抓取相关信号呢？

poemonsense, 2021-12-23:
使用--dump-wave选项会打开波形。-b N -e M会控制只打印在N-M cycle之间的波形

pengshihan10, 2022-01-04:
问题已经解决了，感谢各位同学解答

closed 2022-01-04 by poemonsense