OpenXiangShan/XiangShan #1386: 上板调试遇到问题

pengshihan10, 2021-12-22:
目前正在对“5c7674fe”版本香山处理器与Intel Stratix 10 SX FPGA进行适配。

在上板调试阶段中，从SignalTap调试工具抓取的数据看出，xstop_0_io_mem_axi4_master_araddr信号是从设置的复位起始地址40000000开始，在arvalid信号拉高的时候，以+40h依次读出5个地址直至40000100，且读取的数据与bbl写入存储中地址相对应的数据完全一致，但是之后很多很多个周期都没有访存操作了。

请问一下这个问题是由什么导致的呢？

poemonsense, 2021-12-22:
因为没有更多的有效信息了，我们无法提供帮助。最好是能有波形或其他类型的调试信息，这样我们能帮忙进行调试

pengshihan10, 2021-12-22:
[image]
[image]
[image]
波形图已上传

poemonsense, 2021-12-22:
因为没有cpu内部的波形情况，所以我们也没办法猜到是什么原因造成的。仿真环境下能跑过吗？

poemonsense, 2021-12-22:
如果是把4000_0000设置为起始地址，似乎需要修改PMA的配置，使得指令能够被取出。具体请 @Lemover 确认一下

pengshihan10, 2021-12-22:
目前没有在仿真环境下尝试过。确实修改了PMA的配置，在PMA模块中重新配置了MEMMAP。

pengshihan10, 2021-12-22:
目前没有在仿真环境下尝试过。由于因为修改了香山处理器的地址空间配置，不知道如何对重新配置了的香山处理器进行仿真。

pengshihan10, 2021-12-23:
现在经过修改，我们已经能继续访问到后续地址了，而且arid和rid也是完全匹配的。

但是由于不知道是不是香山Core乱序执行的原因，地址跳转非常杂乱，无法根据生成的bbl文件来确定地址跳转的正确性，甚至执行到后面已经读到了不存在与bbl文件里面的地址和数据。

想请问一下，这种情况是由什么原因导致的呢？

[image]

poemonsense, 2021-12-23:
这个就比较难定位了。建议是，一方面通过仿真来确认一下软件是否是对的，另一方面可以抓一下rob/roq模块的io*commits*valid和io*commits*walk信号，看看是否有指令在提交

pengshihan10, 2021-12-23:
仿真出来的最终结果应该是正确的。看“有指令在提交”是指的是看是否有类似于j跳转这种汇编指令在提交吗？

poemonsense, 2021-12-23:
嗯，是的，如果是一直在往外发取指的话，估计cpu是正常在运行的。接下来就是确认软件是否正确，以及cpu是否正确在执行指令了

pengshihan10, 2022-01-04:
问题已经解决了，感谢各位同学解答

closed 2022-01-04 by poemonsense