OpenXiangShan/XiangShan #1413: 请问调试香山处理器需要什么型号fpga开发板，或者是有什么资源要求

dujiaming1107, 2022-01-01:


poemonsense, 2022-01-03:
默认配置的香山在vu19p上占用大概40-50%的lut资源，可以参考选用fpga板

cisen, 2022-01-19:
> 默认配置的香山在vu19p上占用大概40-50%的lut资源，可以参考选用fpga板

这就已经这么恐怖了吗？玩不起啊，要不再搞个开源vu19p板子，再配合二手芯片估计才能玩得起

poemonsense, 2022-01-19:
> > 默认配置的香山在vu19p上占用大概40-50%的lut资源，可以参考选用fpga板
> 
> 这就已经这么恐怖了吗？玩不起啊，要不再搞个开源vu19p板子，再配合二手芯片估计才能玩得起

可以试一下MinimalConfig配置的香山，应该是比较小的板子就能跑起来。具体资源使用量，我们暂时还没有数据。等香山第二版（南湖）稳定之后，我们统一给一组数据

cisen, 2022-01-19:
> > > 默认配置的香山在vu19p上占用大概40-50%的lut资源，可以参考选用fpga板
> > 
> > 
> > 这就已经这么恐怖了吗？玩不起啊，要不再搞个开源vu19p板子，再配合二手芯片估计才能玩得起
> 
> 可以试一下MinimalConfig配置的香山，应该是比较小的板子就能跑起来。具体资源使用量，我们暂时还没有数据。等香山第二版（南湖）稳定之后，我们统一给一组数据

赞，坐等

aren0924, 2022-02-09:
有参考MinimalConfig的 FPGA 工程的 参考 文档或者 设计 吗 ？？

aren0924, 2022-02-11:
原生的 仿真环境，太多的 device model 都是 DPI-C 实现的。
如果需要往FPGA 上 移植 ， 需要实现这些外设 。。。。。
能不能  在 make verilog 阶段 给一个 for_fpga 的最小系统的 配置。可能只需要一个Flash外设 和 一个串口，如果能生成一个 FPGATop.v 是最好的。

poemonsense, 2022-02-11:
soc您可以参考您的FPGA开发板提供的外设IP来搭建

eatcosmos, 2022-03-12:
> > 默认配置的香山在vu19p上占用大概40-50%的lut资源，可以参考选用fpga板
> 
> 这就已经这么恐怖了吗？玩不起啊，要不再搞个开源vu19p板子，再配合二手芯片估计才能玩得起

国产硬件要崛起啊，外国的fpga价格简直是打劫。

dujiaming1107, 2022-04-02:
> 默认配置的香山在vu19p上占用大概40-50%的lut资源，可以参考选用fpga板

请问你们用的开发板型号是什么啊？

poemonsense, 2022-04-02:
我们用的S2C的vu19p的板子

hyperswine, 2022-04-25:
> 可以试一下MinimalConfig配置的香山，应该是比较小的板子就能跑起来。具体资源使用量，我们暂时还没有数据。等香山第二版（南湖）稳定之后，我们统一给一组数据

Can you give me the specs of minimalconfig? So is there less L1/L2 cache?

AugustusWillisWang, 2022-04-25:
See https://github.com/OpenXiangShan/XiangShan/blob/188f739de96af363761c0f2b80b95b70ad01e0fc/src/main/scala/top/Configs.scala#L54-L172

closed 2022-11-23 by du-jiaming