OpenXiangShan/XiangShan #1427: PRINTF_COND里面的那些debug信息，如何让他们打印出来？

trueif, 2022-01-14:
我就看到了fwrite，没看到fopen呀。

  781     `ifndef SYNTHESIS
  782     `ifdef PRINTF_COND
  783       if (`PRINTF_COND) begin
  784     `endif
  785         if (_T_21 & _T_23) begin
  786           $fwrite(32'h80000002,
  787             "[IFUPC] pc:%x pcUpdate:%d npc:%x RedValid:%d RedTarget:%x LJL:%d LJTarget:%x LJ:%d snpc:%x bpValid:%d pnpn:%x \n"
  788             ,pc,pcUpdate,npc,io_redirect_valid,io_redirect_target,crosslineJumpLatch,crosslineJumpTarget,
  789             bp1_io_crosslineJump,snpc,bp1_io_out_valid,pnpc); // @[Debug.scala 57:13]
  790         end
  791     `ifdef PRINTF_COND
  792       end
  793     `endif
  794     `endif // SYNTHESIS

我看到make emu时define了PRINTF_COND。

poemonsense, 2022-01-14:
您好，可以参考https://www.codenong.com/cs105957818/

hyf6661669, 2022-01-18:
$fwrite直接输出在终端里，$fopen + $fdisplay会将信息打印到指定的文件里

trueif, 2022-01-18:
> 您好，可以参考https://www.codenong.com/cs105957818/

谢谢，有帮助。但是我现在还是没法让他们进STD_ERR。
我修改了scala，把Debug改成了printf，就能在STD_ERR抓到了。

trueif, 2022-01-18:
> $fwrite直接输出在终端里，$fopen + $fdisplay会将信息打印到指定的文件里

Chisel编译的话，不给我机会把fwrite改为fopen+fdisplay

closed 2023-02-15 by AugustusWillisWang