OpenXiangShan/XiangShan #1452: make emu 错误

caohuajia, 2022-02-02:
我可以make verilog,但是make emu报错了,各种工具应该是最新版本,有没有小伙伴有相同的问题并且解决了
```
make -C ./difftest emu SIM_TOP=SimTop DESIGN_DIR=/home/chj/my-project/code/verilog/XiangShan NUM_CORES=1
make[1]: 进入目录“/home/chj/my-project/code/verilog/XiangShan/difftest”
which: no ccache in (/home/chj/.local/bin:/usr/local/bin:/usr/bin:/var/lib/snapd/snap/bin:/usr/local/sbin:/opt/android-sdk/tools:/opt/android-sdk/tools/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl)
make -C /home/chj/my-project/code/verilog/XiangShan sim-verilog
make[2]: 进入目录“/home/chj/my-project/code/verilog/XiangShan”
mkdir -p build
Wed, 02 Feb 2022 20:01:48 +0800
time -a -o ./build/time.log mill -i XiangShan.test.runMain top.SimTop -td build      \
	--config DefaultConfig --full-stacktrace --output-file SimTop.v    \
	--infer-rw --repl-seq-mem -c:top.SimTop:-o:build/SimTop.v.conf  \
	--gen-mem-verilog full --num-cores 1             \
	--enable-difftest
make[2]: time: 没有那个文件或目录
make[2]: *** [Makefile:79：build/SimTop.v] 错误 127
make[2]: 离开目录“/home/chj/my-project/code/verilog/XiangShan”
make[1]: *** [Makefile:29：/home/chj/my-project/code/verilog/XiangShan/build/SimTop.v] 错误 2
make[1]: 离开目录“/home/chj/my-project/code/verilog/XiangShan/difftest”
make: *** [Makefile:111：emu] 错误 2

```

Lingrui98, 2022-02-02:
你需要安装time程序，它在make emu过程中用来统计时间

caohuajia, 2022-02-02:
> 你需要安装time程序，它在make emu过程中用来统计时间

是的,多谢了.一开始是安装了的,不知道为什么会出现问题,又重新安装了一次,现在可以了

closed 2022-03-09 by AugustusWillisWang