OpenXiangShan/XiangShan #1453: 吐槽:真是内存大户啊

caohuajia, 2022-02-02:
为了跑这个工程,我自己的笔记本给配了64g swap,基本全能吃完. cpu占用率却不高,不超过20%. make verilog的3个小时java进程磁盘读取总计1T左右

caohuajia, 2022-02-02:
[image]

ljwljwljwljw, 2022-02-02:
我们在下个版本中会添加 [CIRCT](https://github.com/llvm/circt) 编译支持，可以解决编译速度慢和内存占用高的问题

sequencer, 2022-02-07:
💰

pascalApple, 2022-02-16:
我后来为了跑香山捡了个96G内存的双路至强配寨板的洋垃圾

trueif, 2022-02-24:
用了楼主的方法，内存16GB的机器，把swap搞成64GB，跑通了……不容易。香山好大，仿真比果壳慢了好几倍。

acclea, 2022-07-01:
可以把内存控制到16G么，甚至8G么，小家小户没那么多的内存

trueif, 2022-07-01:
楼上的朋友，16GB内存可以跑。把SWAP扩大，SWAP只需要硬盘够就行。

acclea, 2022-07-06:
> 
如果直接外接一个移动硬盘，用移动硬盘可以么，机器本身的硬盘没多少

closed 2023-10-21 by cebarobot

poemonsense, 2023-10-21:
已经支持CIRCT编译，内存占用极低

sequencer, 2023-10-21:
放心 builder里的chiselir还占一半的内存

jerryhethatday, 2023-10-21:
> 已经支持CIRCT编译，内存占用极低

16gb能跑吗？

poemonsense, 2023-10-21:
> > 已经支持CIRCT编译，内存占用极低
> 
> 16gb能跑吗？

没有升级CIRCT之前16G也能跑