OpenXiangShan/XiangShan #1459: 产生 波形

aren0924, 2022-02-09:
make emu -j32 之后产生默认配置的香山核 ， 跑仿真如何产生波形文件呢 ？
比如 --dump-wave -b 0 -e 1100 ， ./build 目录下没有  vcd 文件

poemonsense, 2022-02-09:
我们这边复现不了这个问题。您可以检查一下NOOP_HOME的环境变量，是否指向的是XiangShan这个目录，波形默认会生成在$NOOP_HOME/build目录下

aren0924, 2022-02-09:
是不是 需要  make emu EMU_TRACE=1  -j32 ??
NOOP_HOME 的设定 肯定没有问题的

aren0924, 2022-02-09:
如果用  make emu EMU_TRACE=1 编出来的 emu 跑 ， 会有错误 , 但是 确实有dump wave-form的提示
aren@aren-p15:~/works/Github/XiangShan/xs-env/XiangShan$ ./build/emu -i ./ready-to-run/linux.bin --dump-wave -b 0 -e 10000
Emu compiled at Feb  9 2022, 10:28:10
Assertion failed at line 957970.
Assertion failed at line 956459.
Assertion failed at line 959340.
Assertion failed at line 959388.
Assertion failed at line 959412.
Assertion failed at line 959388.
Assertion failed at line 959340.
Assertion failed at line 959388.
Assertion failed at line 959340.
Assertion failed at line 959388.
Assertion failed at line 959412.
Assertion failed at line 959340.
Assertion failed at line 959364.
Assertion failed at line 959412.
Assertion failed at line 959340.
Assertion failed at line 959388.
.........

The image is ./ready-to-run/linux.bin
Using simulated 8192MB RAM

**dump wave to /home/aren/works/Github/XiangShan/xs-env/XiangShan/build/2022-02-09@11:46:08.vcd...**

--diff is not given, try to use $(NEMU_HOME)/build/riscv64-nemu-interpreter-so by default
NemuProxy using /home/aren/works/Github/XiangShan/xs-env/NEMU/build/riscv64-nemu-interpreter-so
[src/device/io/mmio.c,18,add_mmio_map] Add mmio map 'flash' at [0x0000000010000000, 0x00000000100000ff]
The simulation stopped. There might be some assertion failed.
Core 0: ABORT at pc = 0x0
total guest instructions = 0
instrCnt = 0, cycleCnt = 0, IPC = -nan
Seed=0 Guest cycle spent: 0 (this will be different from cycleCnt if emu loads a snapshot)
Host time spent: 5ms
Assertion failed: SinkD must receive aligned message when needData
    at SinkD.scala:46 assert(!io.d.valid || !needData || io.d.bits.size === log2Up(blockBytes).U, "SinkD must receive aligned message when needData")
Assertion failed: Only data task can go to stage3!
    at SourceD.scala:206 assert(!s3_valid || needData(s3_regs.req), "Only data task can go to stage3!")
Assertion failed: Directory result was sent to mismatch MSHR(mshrId:15, resultId:15)
    at MSHR.scala:48 assert(
Assertion failed
    at MSHR.scala:145 assert(RegNext(!meta_valid || !req.fromC || meta.hit, true.B)) // Release should always hit
Assertion failed
    at MSHR.scala:575 assert(RegNext(!req_valid || !io.alloc.valid, true.B)) // TODO: support fully-pipelined
Assertion failed
    at MSHR.scala:145 assert(RegNext(!meta_valid || !req.fromC || meta.hit, true.B)) // Release should always hit
Assertion failed: Directory result was sent to mismatch MSHR(mshrId:13, resultId:15)
    at MSHR.scala:48 assert(
Assertion failed
    at MSHR.scala:145 assert(RegNext(!meta_valid || !req.fromC || meta.hit, true.B)) // Release should always hit
Assertion failed: Directory result was sent to mismatch MSHR(mshrId:12, resultId:15)
    at MSHR.scala:48 assert(
Assertion failed
    at MSHR.scala:145 assert(RegNext(!meta_valid || !req.fromC || meta.hit, true.B)) // Release should always hit
Assertion failed
 .......

所以，是不是 需要 增加其他的 编译选项  ？？？ 
谢谢

poemonsense, 2022-02-09:
波形会生成在/home/aren/works/Github/XiangShan/xs-env/XiangShan/build/2022-02-09@11:46:08.vcd这个文件，和log里面说的一样

poemonsense, 2022-02-09:
嗯，需要打开EMU_TRACE=1

aren0924, 2022-02-09:
没有波形，用 make emu -EMU_TRACE=1 编出来的 emu ,跑起来就因为  assertion failed ， 中断了。
但是不加 -EMU_TRACE=1 的 就可以跑。
大概是啥问题呢 ？

aren0924, 2022-02-09:
重新 clone了一个 XiangShan 目录， 然后 重新编译了一下，就没有问题了。产生了的波形用gtkwave ， 可以打开。。。

closed 2022-03-09 by AugustusWillisWang