OpenXiangShan/XiangShan #1460: difftest

aren0924, 2022-02-09:
difftest 的目的是用 其他仿真工具 来作同样的测试 ，是吧 ？？
看到 difftest 目录下 有 vcs.mk , 请问怎么能用  vcs  flow 跑起来呢 ？有没有文档 说明的 ， 谢谢拉

poemonsense, 2022-02-09:
make simv就可以，其他选项可以参考vcs.mk文件的具体内容

aren0924, 2022-02-09:
恩，有questa flow 的 .mk 吗 ？？

poemonsense, 2022-02-09:
您可以参考现有的makefile改一改，我们不（官方地）维护在商用EDA软件上的flow。

aren0924, 2022-02-09:
嗯，好的。
但是， VCS 不是也是 Synopsis的商用EDA软件。

poemonsense, 2022-02-09:
我们不保证vcs.mk能够正常工作

aren0924, 2022-02-09:
恩，理解。

另一个问题，香山有在FPGA 平台上 Bring-Up的指导和经验分享吗 ？ 
这边是有 VU19P的FPGA 平台，希望能Bring-Up一下。

poemonsense, 2022-02-09:
XSTop出来是一个外设、一个memory和一个DMA通道。可以根据您那边FPGA外设环境来包一个soc环境，就可以跑

aren0924, 2022-02-09:
恩， 谢谢。。。

aren0924, 2022-02-09:
make simv -n 后 得到的 详细编译指令是：
vcs -full64 +v2k -timescale=1ns/1ns -sverilog -debug_access+all +lint=TFIPC-L +vcs+initreg+random -CFLAGS "-std=c++11 -static -Wall -I/home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/vcs -I/home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common -I/home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/difftest -I/home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/plugin/include" -LDFLAGS "-lpthread -lSDL2 -ldl -lz" -j200 -y /home/aren/works/Github/XiangShan/XiangShan/build +libext+.v -Mdir=/home/aren/works/Github/XiangShan/XiangShan/build/simv-compile ../build/SimTop.v /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/axi4.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/remote_bitbang.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/flash.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/compress.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/lightsss.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/ram.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/SimJTAG.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/keyboard.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/common.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/vga.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/uart.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/sdcard.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/device.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/difftest/ref.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/difftest/interface.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/difftest/difftest.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/difftest/goldenmem.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/difftest/refproxy.cpp /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/vcs/main.cpp ./src/test/vsrc/common/EICG_wrapper.v ./src/test/vsrc/common/SimJTAG.v ./src/test/vsrc/common/ram.v ./src/test/vsrc/common/ref.v ./src/test/vsrc/common/assert.v /home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/vsrc/vcs/top.v

这个就是 VCS flow的 makefile , 是吧 ？

生成 simv  之后 ， ./simv -workload=xxx.bin 就可以了是吧 ？

aren0924, 2022-02-11:
作了一个 Questa的仿真工程 ， 基本功能是对的。

aren0924, 2022-02-11:
关于 $AM_HOME 目录下 的那些 app test case ,  都可以 用 make emu CONFIG=MinimalConfig EMU_TRACE=1 -j32 的编译命令产生的 emu 来测试 吗 ？
我想 试一试  typing 的  case , 需要额外的 配置 编译参数 吗 ？

poemonsense, 2022-02-11:
嗯，都可以用编译出来的emu来测试。编译命令应该是make ARCH=riscv64-xs，默认应该是RV64G，O2的选项

aren0924, 2022-02-14:
需要在 ./difftets/build/config/config.h 里，需要打开类似 SHOW_SCREEN这样的 定义 吗 ??

aren0924, 2022-02-14:
我打开 这个#define SHOW_SCREEN, 然后 make emu CONFIG=MinimalConfig EMU_TRACE=1 -j32 。 
编译报错.
home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/device.cpp:47:3: error: unknown type name 'SDL_Event'
  SDL_Event event;
  ^
/home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/device.cpp:56:62: error: use of undeclared identifier 'SDL_KEYDOWN'
                        bool is_keydown = (event.key.type == SDL_KEYDOWN);
                                                             ^
/home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/device.cpp:50:12: error: use of undeclared identifier 'SDL_QUIT'
      case SDL_QUIT: break; //set_abort();
           ^
/home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/device.cpp:53:12: error: use of undeclared identifier 'SDL_KEYDOWN'
      case SDL_KEYDOWN:
           ^
/home/aren/works/Github/XiangShan/XiangShan/difftest/src/test/csrc/common/device.cpp:54:12: error: use of undeclared identifier 'SDL_KEYUP'
      case SDL_KEYUP: {

poemonsense, 2022-02-14:
> 需要在 ./difftets/build/config/config.h 里，需要打开类似 SHOW_SCREEN这样的 定义 吗 ??

不需要打开

aren0924, 2022-02-14:
我是这么 跑 ./build/emu -i xxxx/xxxx/XiangShan/xs-env/nexus-am/apps/typing/build/typing-riscv64-xs.bin 
但是 ， 没有 没有什么交互信息显示。只有类似 
now = 456s
now = 517s
now = 587s 
....
...
这样的信息 

是怎么一回事儿呢 ？？  谢谢

poemonsense, 2022-02-14:
这是通过VGA来连接的一个带有图形界面的小应用。建议您可以先试一下coremark, microbench这些不带图形界面的测试

aren0924, 2022-02-14:
所有 build 目录下的  DifftestXXX.v 如果都变成 dummy模块可以吗 ？ 比如我在作仿真的时候 ， 定义 +define+SYNTHESIS  , 会影响功能 吗 ？
谢谢

poemonsense, 2022-02-14:
Difftest*模块是给difftest检查使用的，可以在仿真的时候使用`--no-diff`来关闭检查

aren0924, 2022-02-15:
我之前 bring-up了 一个 Questa-flow 的仿真工程。简单的只是跑 hello world 的打印。
一开始可能是因为，打印的内容不多，所以以为功能是正确的。
但是，当我把打印输出的内容变长，变多的时候，打印到中间位置，就没有输出了 。 像是 CPU 取指令 出错或者 停下来了。
参考的是在 difftest 目录下执行 make simv 对应的 filelist 和 参数. 

请问，VCS-flow 是经过验证，完全可以跑的吗 ？  谢谢

poemonsense, 2022-02-15:
如果能跑的话，说明flow是没问题的，毕竟用的是同一套verilog代码。VCS flow不保证所有功能都是对的

aren0924, 2022-02-15:
用的是同一套代码 。
问题是 Chisel 生成的 verilog 代码，调试是个问题。
能不能提供一个 signal list ,用来帮助 / 方便 调试 这种 功能问题的？？ 或者设计内部有没有 类似 状态寄存器 或者 状态信号，可以给出一些定位问题的信息和 指导。
谢谢 ！

poemonsense, 2022-02-15:
我们目前也没有这样的列表。我们一般在调试的时候，会优先关注io_*这样的信号，一般都是模块的io信号

poemonsense, 2022-02-15:
您可以先试试在verilator环境下对应的case能不能跑，一般rtl是没有问题的，估计是环境或者case本身的问题

aren0924, 2022-02-16:
是的，我想顺着CPU debug的思路追踪一下，通过有波形，以及hello 程序的反汇编，应该可以看到CPU的行为。为啥会停在一条指令就不动了 。。。

香山 Core内部类似PC 指针这样的CPU信息，可以去哪些模块获取呢 ？

谢谢。。。

poemonsense, 2022-02-16:
可以优先关注一下CtrlBlock/ROB/io_commits_*相关信号，里面有PC和指令提交的信息

aren0924, 2022-02-16:
one quick question ,  
在编译的阶段 ， 能不能 整体 disable difftest 功能 ？

poemonsense, 2022-02-17:
如果需要完全关闭，您可以将https://github.com/OpenXiangShan/XiangShan/blob/c7fabd05bd7e19d09ab8dcb2c824ab5423020660/src/main/scala/xiangshan/Parameters.scala#L266这一行改成AlwaysBasicDiff: Boolean = false，同时在make的时候添加RELEASE=1，就会完全关闭difftest功能。或者在运行的时候--no-diff也可以。difftest本身不会对编译产生开销，且不会对RTL仿真产生任何影响，不影响原有电路的运行

aren0924, 2022-02-17:
遇到的问题是这样的， 重新bring-up 了Questa 仿真，g++用的 Questa的安装版本，这是与原来不同的。
SIM 1> run 15us
# [src/device/io/mmio.c,18,add_mmio_map] Add mmio map 'flash' at [0x0000000010000000, 0x00000000100000ff]
# simv compiled at Feb 17 2022, 08:24:04
# --diff is not given, try to use $(NEMU_HOME)/build/riscv64-nemu-interpreter-so by default
# NemuProxy using /home/emu/works/XiangShan/build/riscv64-nemu-interpreter-so
# The image is ./hello-riscv64-xs.bin
# Using simulated 8192MB RAM
#  >>>> Hello Xiangshan! <<<< No instruction commits for 5000 cycles of core 0. Please check the first instruction.
# Note: The first instruction may lie in 0x10000000 which may executes and commits after 500 cycles.
#    Or the first instruction may lie in 0x80000000 which may executes and commits after 2000 cycles.
# ** Fatal: (vsim-12005) Undefined function 'test_spike()' introduced from './basic.so' is being called. Exiting ... 
# End time: 08:24:59 on Feb 17,2022, Elapsed time: 0:00:52
# Errors: 1, Warnings: 5

仿真 runtime的 参数是这样的 vsim -c opt_top -sv_lib basic +workload=./hello-riscv64-xs.bin ， 类似 emu -i ./hello-rsic64-xs.bin 的话，如何传递 --no-diff 参数 ？？ 谢谢

poemonsense, 2022-02-17:
这个需要您在VCS仿真的top.v里面修改一下，把difftest相关的调用注释掉即可，比如difftest_step的调用。

对于上面这个hello的case，看起来应该是正确跑完了。因为我们现在是通过difftest来判断是否结束的，所以关闭difftest后，确实就没法确认程序是否退出了，应该是后面已经跑飞了

aren0924, 2022-02-17:
Emu compiled at Feb 14 2022, 21:43:19
The image is /home/aren/works/Github/XiangShan/xs-env/nexus-am/apps/hello/build/hello-riscv64-xs.bin
Using simulated 8192MB RAM
--diff is not given, try to use $(NEMU_HOME)/build/riscv64-nemu-interpreter-so by default
NemuProxy using /home/aren/works/Github/XiangShan/xs-env/NEMU/build/riscv64-nemu-interpreter-so
[src/device/io/mmio.c,18,add_mmio_map] Add mmio map 'flash' at [0x0000000010000000, 0x00000000100000ff]
The first instruction of core 0 has commited. Difftest enabled. 
 >>>> Hello Xiangshan! <<<< 3600 
 >>> XiangShan GoodBye ! <<<< 
Core 0: HIT GOOD TRAP at pc = 0x80000068
total guest instructions = 2,312

正常的打印是这样的，后面还有一段的。上面的那个 hello 的 case 其实没有跑完。。。。

poemonsense, 2022-02-17:
看懂了。上面的那个case估计是5000拍的时候，发现一直没有指令提交，就直接退出了。您可以把top.v里面difftest_step相关的注释掉，应该就可以了。

aren0924, 2022-02-18:
启动 ready-to-run / linux.bin 的 话 ，对ram 外设有什么要求吗  ? verilator的仿真的配置是 8G ， 如果在其他配置的机器上 ，没有8G 的内存，还可以启动吗 ？？谢谢

poemonsense, 2022-02-18:
linux.bin应该是256MB内存就能够启动了

poemonsense, 2022-02-21:
linux.bin的启动是会打印出Hello World。linux.bin应该需要的内存数量比较小，您可以再试试512M或者1G的选项。另外，linux.bin实际使用到的内存只有几十M-几百M。

aren0924, 2022-02-22:
我在Questa-flow里启动linux.bin，开了8G的内存，但是difftest的部分注释掉了。
启动过程能够打印到  R 和V的 log  之后 , 停在那边了, INSTRUCTION SETS WANT TO BE  FREE 之后就不动了。。。。
这个过程是 BootLoader 启动 kernel刚开始的地方吧， 是不是什么外设没有设对 ?? 。。。
正常的verilator 启动，看到的是 OF： fdt : Ignoring memory range 0x80000000 - 0x80200000

poemonsense, 2022-02-23:
这里可能不是停住了，是仿真速度比较慢，这里应该确实需要一段时间的

closed 2023-02-15 by AugustusWillisWang