OpenXiangShan/XiangShan #1471: Betapoint

wanermian, 2022-02-19:
请问Betapoint的相关介绍除了去年risc-v中国峰会上的分享后续还有更新介绍吗？
主要想了解下怎么把functional warmup得到的状态导入到simulator模型里的，另外除了cache状态还有哪些关键状态需要导入？
谢谢。

shinezyy, 2022-02-21:
我们暂时没有更新。
我们有一个比较脏的方案，但是这个方案需要在RTL改变之后也手动修改C++的simualtor。
我们过去几个月尝试了几种干净的、自动化的方案，都失败了。
目前我们还在寻找其他解决方案。

关于如何functional warmup得到的状态导入到simulator：Verilog用readmemh就行了，Chisel有对readmemh的封装。
关于需要导入哪些状态，我们目前看到的比较重要的是Cache、BTB 和 分支预测器的状态。

wanermian, 2022-02-21:
> 我们暂时没有更新。 我们有一个比较脏的方案，但是这个方案需要在RTL改变之后也手动修改C++的simualtor。 我们过去几个月尝试了几种干净的、自动化的方案，都失败了。 目前我们还在寻找其他解决方案。
> 
> 关于如何functional warmup得到的状态导入到simulator：Verilog用readmemh就行了，Chisel有对readmemh的封装。 关于需要导入哪些状态，我们目前看到的比较重要的是Cache、BTB 和 分支预测器的状态。

你们有对比过在verilator上仿真50M warmup+50M采样得到的性能和视频里介绍的前面用NEMU functional warmup，verilator只跑50K指令的两种方式的所得到的SPEC性能差异吗？和现在芯片上跑出来的实际性能比，误差分别是多少呢？

shinezyy, 2022-02-21:
verilator上仿真50M warmup+50M采样 VS. FPGA VS. 真实芯片的数据在这里：https://weibo.com/1686707751/Lg6de4uQi

Functional warmup的性能数据暂时还没有，因为目前我们的functional warmup 基础设施还不完善。我们一致致力于寻找干净的方案，并没有用脏的方案跑过完整的性能数据。

wanermian, 2022-02-21:
[image]
非常感谢。
另外还有个问题想请教下，在上图中的NEMU fast-forwarding，NEMU functional warmup，verilator warmup阶段分别都做了哪些事情呢？

shinezyy, 2022-02-21:
NEMU fast-forwarding是不带functional Cache的，目的是跳过尽可能多的指令；
NEMU functional warmup带functional Cache；
verilator warmup是用真实的RTL来进一步warmup cache和流水级；

wanermian, 2022-02-21:
> NEMU fast-forwarding是不带functional Cache的，目的是跳过尽可能多的指令； NEMU functional warmup带functional Cache； verilator warmup是用真实的RTL来进一步warmup cache和流水级；

了解了，多谢。期待你们的后续更新。

wanermian, 2022-03-01:
不好意思，再请教个问题。
还是以上图为例，切出来的20亿条指令，分了四段去执行，那这20亿条指令片段总的性能要怎么计算呢？

shinezyy, 2022-03-01:
用Verilator Simulation（红色部分）的IPC替代这一段的IPC，认为他在这一段的性能是不变的

wanermian, 2022-03-01:
> 用Verilator Simulation（红色部分）的IPC替代这一段的IPC，认为他在这一段的性能是不变的

verilator是50K指令，基于什么方面判断和total 20亿的性能一致呢？是对两部分代码段做了什么统计吗？

shinezyy, 2022-03-01:
SMARTS采样就是这样的，没有统计，所以可能有误差。