OpenXiangShan/XiangShan #1472: Waveform terminator?

cnjsdfcy, 2022-02-22:
Hi,

XiangShan真是很棒的开源项目！配套的工具也都很实用。
从分享的文档中有看到，很多辅助工具没有开放出来，比如Waveform terminator。
请问香山团队有计划开源这些辅助工具吗？

Thanks~

poemonsense, 2022-02-22:
我们的工具都是全部开源的，可以请 @ljwljwljwljw 介绍一下这个工具各个部分的开源地址

ljwljwljwljw, 2022-02-23:
这个工具也是开源的:(https://github.com/OpenXiangShan/vcd-to-log)
但目前遇到的问题是vcd解析很慢，对于大波形还是很难用

cnjsdfcy, 2022-02-23:
Nice! 后面有机会看能不能一起优化加速一下。
另外，performance verification相关的tools，比如下图中提到的：GCPT, Betapoint, KFDB，是否也都是开源的呀？我翻了一下貌似没有找到~
[image]

poemonsense, 2022-02-23:
GCPT是开源的，具体代码应该是在NEMU仓库下面，具体请 @shinezyy 帮忙确定
Betapoint相关的可以参考 https://github.com/OpenXiangShan/XiangShan/issues/1471
KFDB也是开源的，在https://github.com/OpenXiangShan/perfUI，请 @wakafa1 确定

wakafa1, 2022-02-23:
https://github.com/OpenXiangShan/perfUI 提供了性能计数器可视化工具，项目使用 QT 编写，后续还将进行改进

shinezyy, 2022-02-23:
GCPT生成可以看这里：
https://github.com/OpenXiangShan/XiangShan-doc/blob/main/tutorial/others/Checkpoint%E7%9A%84%E7%94%9F%E6%88%90.md

cnjsdfcy, 2022-02-23:
Great！太棒了~下来好好研究一下。
多谢~

closed 2022-02-23 by cnjsdfcy