OpenXiangShan/XiangShan #1491: The latest version of XiangShan,包括哪些？

gms-1948, 2022-03-14:
请列出明细，谢谢！

poemonsense, 2022-03-14:
不好意思，我们没有理解您的问题。

gms-1948, 2022-03-14:
make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_TRACE=1 -j32无法通过。
你们回答：Please use the latest version of XiangShan, update all submodules, and try again.
我：git submodule update --init --recursive
       make emu ... 报错和原来一样。
问题：还要更新什么？

poemonsense, 2022-03-14:
same as #1487 #1489

poemonsense, 2022-03-14:
@AugustusWillisWang need to bump xs-env?

poemonsense, 2022-03-14:
> make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_TRACE=1 -j32无法通过。 你们回答：Please use the latest version of XiangShan, update all submodules, and try again. 我：git submodule update --init --recursive make emu ... 报错和原来一样。 问题：还要更新什么？

您可以试一下更新到最新版本的香山
```
cd XiangShan
git pull
git checkout origin/master
```

AugustusWillisWang, 2022-03-14:
如果更新后还是有问题, 可以 `git log` 一下香山当前的 commit id 给我们

gms-1948, 2022-03-16:
> 
我的Verilator 4.028 按你的要求需4.204，如何升级？还是卸载重装？

gms-1948, 2022-03-16:
> 

> 如果更新后还是有问题, 可以 `git log` 一下香山当前的 commit id 给我们
commit 184a195889c871432da1d39e2885d0af812be3fe (HEAD)
Author: zfw <31198653+fwzhang@users.noreply.github.com>
Date:   Thu Aug 26 12:20:30 2021 +0800

AugustusWillisWang, 2022-03-16:
> commit [184a195](https://github.com/OpenXiangShan/XiangShan/commit/184a195889c871432da1d39e2885d0af812be3fe) (HEAD)
> Author: zfw [31198653+fwzhang@users.noreply.github.com](mailto:31198653+fwzhang@users.noreply.github.com)
> Date:   **Thu Aug 26 12:20:30 2021 +0800**

您使用的 commit 已经非常老了, 请按这里的说明更新到最新版本:

> 
> 您可以试一下更新到最新版本的香山
> 
> ```
> cd XiangShan
> git pull
> git checkout origin/master
> ```

> 我的Verilator 4.028 按你的要求需4.204，如何升级？还是卸载重装？

gms-1948, 2022-03-28:
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# git pull
fatal: unable to access 'https://github.com/OpenXiangShan/XiangShan/': GnuTLS recv error (-54): Error in the pull function.

Lingrui98, 2022-03-28:
> root@gms-VirtualBox:/home/gms/xs-env/XiangShan# git pull
> fatal: unable to access 'https://github.com/OpenXiangShan/XiangShan/': GnuTLS recv error (-54): Error in the pull function.

请检查您的网络是否连通到GitHub

gms-1948, 2022-03-30:
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# verilator --version
Verilator 4.218 2022-01-17 rev v4.218
更新后
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# git pull
fatal: unable to access 'https://github.com/OpenXiangShan/XiangShan/': GnuTLS recv error (-54): Error in the pull function.

gms-1948, 2022-03-31:
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# git branch
* (HEAD detached at b6c4f5c1a)
  master
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# git pull master
fatal: 'master' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
请指导！

gms-1948, 2022-03-31:
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# cat .__head__
//commit e41db104933c1b3efe17b37ce1d6ca0f65d7e877
//Author: happy-lx <54952983+happy-lx@users.noreply.github.com>
//Date:   Sun Mar 27 10:18:20 2022 +0800
//
//    sq: fix use of OHToUInt (#1505)

poemonsense, 2022-03-31:
现在的问题是啥？编译不了emu？

gms-1948, 2022-04-01:
编译不了emu

poemonsense, 2022-04-01:
出错现场是啥？

gms-1948, 2022-04-01:
我认为是对git管理模式了解问题。
1. 我是通过clone得到XiangShan，版本旧需更新，git pull通不过。
2. 如何才能和master同步？

poemonsense, 2022-04-01:
关于第一个问题，`fatal: unable to access 'https://github.com/OpenXiangShan/XiangShan/': GnuTLS recv error (-54): Error in the pull function.` 这个提示意思是您那边网路环境有问题，无法连接到GitHub

gms-1948, 2022-04-01:
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# git pull
remote: Enumerating objects: 24, done.
remote: Counting objects: 100% (24/24), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 14 (delta 6), reused 14 (delta 6), pack-reused 0
Unpacking objects: 100% (14/14), 1.40 KiB | 34.00 KiB/s, done.
From https://github.com/OpenXiangShan/XiangShan
 * [new branch]          flash-fetch -> origin/flash-fetch
You are not currently on a branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>

gms-1948, 2022-04-01:
切换到master的目的是获取新版文件，如何获取？

gms-1948, 2022-04-01:
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# git checkout origin/master
M	difftest
M	ready-to-run
M	rocket-chip
Previous HEAD position was b6c4f5c1a Merge pull request #1506 from OpenXiangShan/fix-vcs
HEAD is now at 128703a2e Merge pull request #1517 from OpenXiangShan/circt-support
root@gms-VirtualBox:/home/gms/xs-env/XiangShan#

AugustusWillisWang, 2022-04-01:
RTFM
http://git-scm.com/book/zh/v2
http://git-scm.com/book/en/v2

poemonsense, 2022-04-01:
> root@gms-VirtualBox:/home/gms/xs-env/XiangShan# git checkout origin/master M difftest M ready-to-run M rocket-chip Previous HEAD position was [b6c4f5c](https://github.com/OpenXiangShan/XiangShan/commit/b6c4f5c1a375060defe0376679115e7f0b687c81) Merge pull request #1506 from OpenXiangShan/fix-vcs HEAD is now at [128703a](https://github.com/OpenXiangShan/XiangShan/commit/128703a2ef83b5e09cd222c9d53acf2e38acd426) Merge pull request #1517 from OpenXiangShan/circt-support root@gms-VirtualBox:/home/gms/xs-env/XiangShan#

这个提示，意味着您那边已经是最新的master了

gms-1948, 2022-04-01:
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_TRACE=1 -j32
make: the '-j' option requires a positive integer argument
make -C ./difftest emu SIM_TOP=SimTop DESIGN_DIR=/home/gms/xs-env/XiangShan NUM_CORES=1
make[1]: Entering directory '/home/gms/xs-env/XiangShan/difftest'
make -C /home/gms/xs-env/XiangShan sim-verilog
make[2]: Entering directory '/home/gms/xs-env/XiangShan'
mkdir -p build
Fri, 01 Apr 2022 11:26:47 +0800
time -a -o ./build/time.log mill -i XiangShan.test.runMain top.SimTop -td build  \
	--config MinimalConfig                                            \
	--infer-rw --repl-seq-mem -c:top.SimTop:-o:build/SimTop.v.conf --gen-mem-verilog full                                               \
	--num-cores 1                                      \
	--disable-log --enable-difftest 
sed -i -e 's/_\(aw\|ar\|w\|r\|b\)_\(\|bits_\)/_\1/g' build/SimTop.v
sed: can't read build/SimTop.v: No such file or directory
make[2]: *** [Makefile:101: build/SimTop.v] Error 2
make[2]: Leaving directory '/home/gms/xs-env/XiangShan'
make[1]: *** [Makefile:29: /home/gms/xs-env/XiangShan/build/SimTop.v] Error 2
make[1]: Leaving directory '/home/gms/xs-env/XiangShan/difftest'
make: *** [Makefile:139: emu] Error 2
编译emu出错

poemonsense, 2022-04-01:
可以通过以下两步检查您的环境：
`which mill`
`ls build`

gms-1948, 2022-04-01:
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# which mill
/usr/local/bin/mill
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# ls build
firrtl_black_box_resource_files.f  XSTop.0x0.0.regmap.json         XSTop.fir
lock-emu                           XSTop.0x38000000.0.regmap.json  XSTop.graphml
plusarg_reader.v                   XSTop.0x38010000.0.regmap.json  XSTop.json
sim_sram.v                         XSTop.0x38020000.0.regmap.json  XSTop.plusArgs
SimTop.v.sram.v                    XSTop.0x3c000000.0.regmap.json  XSTop.v
time.log                           XSTop.0x40.0.regmap.json        XSTop.v.conf
tsmc28_sram.v                      XSTop.anno.json
tsmc28_sram.v.conf                 XSTop.dts

poemonsense, 2022-04-01:
看起来您那边没有正确地生成仿真程序。可以`rm -rf build`之后，重新`make emu CONFIG=MinimalConfig`

gms-1948, 2022-04-01:
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# rm -rf build
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# make emu CONFIG=MinimalConfig
make: the '-j' option requires a positive integer argument
make -C ./difftest emu SIM_TOP=SimTop DESIGN_DIR=/home/gms/xs-env/XiangShan NUM_CORES=1
make[1]: Entering directory '/home/gms/xs-env/XiangShan/difftest'
make -C /home/gms/xs-env/XiangShan sim-verilog
make[2]: Entering directory '/home/gms/xs-env/XiangShan'
mkdir -p build
Fri, 01 Apr 2022 14:24:06 +0800
time -a -o ./build/time.log mill -i XiangShan.test.runMain top.SimTop -td build  \
	--config MinimalConfig                                            \
	--infer-rw --repl-seq-mem -c:top.SimTop:-o:build/SimTop.v.conf --gen-mem-verilog full                                               \
	--num-cores 1                                      \
	--enable-difftest 
sed -i -e 's/_\(aw\|ar\|w\|r\|b\)_\(\|bits_\)/_\1/g' build/SimTop.v
sed: can't read build/SimTop.v: No such file or directory
make[2]: *** [Makefile:101: build/SimTop.v] Error 2
make[2]: Leaving directory '/home/gms/xs-env/XiangShan'
make[1]: *** [Makefile:29: /home/gms/xs-env/XiangShan/build/SimTop.v] Error 2
make[1]: Leaving directory '/home/gms/xs-env/XiangShan/difftest'
make: *** [Makefile:139: emu] Error 2

poemonsense, 2022-04-01:
`mill -i XiangShan.test.runMain top.SimTop -td build
--config MinimalConfig
--infer-rw --repl-seq-mem -c:top.SimTop:-o:build/SimTop.v.conf --gen-mem-verilog full
--num-cores 1
--enable-difftest` 这个命令没有输出吗？手动运行有输出吗？

gms-1948, 2022-04-01:
手动运行没有输出。

poemonsense, 2022-04-01:
可以试一下先`make init`

gms-1948, 2022-04-02:
root@gms-VirtualBox:/home/gms/xs-env/XiangShan# make emu CONFIG=MinimalConfig
make: the '-j' option requires a positive integer argument
make -C ./difftest emu SIM_TOP=SimTop DESIGN_DIR=/home/gms/xs-env/XiangShan/XiangShan NUM_CORES=1
make[1]: Entering directory '/home/gms/xs-env/XiangShan/difftest'
make -C /home/gms/xs-env/XiangShan/XiangShan sim-verilog
make[2]: Entering directory '/home/gms/xs-env/XiangShan/XiangShan'
make[2]: *** No rule to make target 'sim-verilog'.  Stop.
make[2]: Leaving directory '/home/gms/xs-env/XiangShan/XiangShan'
make[1]: *** [Makefile:34: /home/gms/xs-env/XiangShan/XiangShan/build/SimTop.v] Error 2
make[1]: Leaving directory '/home/gms/xs-env/XiangShan/difftest'
make: *** [Makefile:139: emu] Error 2
重新执行：rm -rf build; make init; make emu CONFIG.....没有通过，build目录没有建成。

poemonsense, 2022-04-02:
请提供完整的log

elta, 2022-06-04:
make verilog编译不通过
-----------------------------

commit 9b4044e7488d7ef8d300a637c9008ce9fb95e0e9 (HEAD -> master, origin/master, origin/HEAD)
Author: Yinan Xu <xuyinan@ict.ac.cn>
Date:   Tue May 31 16:27:55 2022 +0800

    soc: add synchronizers for external interrupt bits (#1566)

----------------------------------------------------------
$ make verilog 
mkdir -p build
time -a -o ./build/time.log mill -i XiangShan.runMain top.TopMain -td build  \
	--config DefaultConfig                                        \
	--infer-rw --repl-seq-mem -c:top.TopMain:-o:build/XSTop.v.conf --gen-mem-verilog full                                          \
	--num-cores 1                                  \
	--disable-all --remove-assert --fpga-platform 
sed -i -e 's/_\(aw\|ar\|w\|r\|b\)_\(\|bits_\)/_\1/g' build/XSTop.v
sed: can't read build/XSTop.v: No such file or directory
make: *** [Makefile:76: build/XSTop.v] Error 2
---------------------------
$ ls build
time.log
$ cat build/time.log 
0.00user 0.00system 0:00.00elapsed 0%CPU (0avgtext+0avgdata 1612maxresident)k
0inputs+0outputs (0major+73minor)pagefaults 0swaps
0.00user 0.00system 0:00.00elapsed 0%CPU (0avgtext+0avgdata 1616maxresident)k
0inputs+0outputs (0major+74minor)pagefaults 0swaps

elta, 2022-06-04:
重新下载mill之后，这一步过去了

> ## make verilog编译不通过
> commit [9b4044e](https://github.com/OpenXiangShan/XiangShan/commit/9b4044e7488d7ef8d300a637c9008ce9fb95e0e9) (HEAD -> master, origin/master, origin/HEAD) Author: Yinan Xu [xuyinan@ict.ac.cn](mailto:xuyinan@ict.ac.cn) Date: Tue May 31 16:27:55 2022 +0800
> 
> ```
> soc: add synchronizers for external interrupt bits (#1566)
> ```
> 
> $ make verilog mkdir -p build time -a -o ./build/time.log mill -i XiangShan.runMain top.TopMain -td build --config DefaultConfig --infer-rw --repl-seq-mem -c:top.TopMain:-o:build/XSTop.v.conf --gen-mem-verilog full --num-cores 1 --disable-all --remove-assert --fpga-platform sed -i -e 's/_(aw|ar|w|r|b)_(|bits_)/_\1/g' build/XSTop.v sed: can't read build/XSTop.v: No such file or directory make: *** [Makefile:76: build/XSTop.v] Error 2
> 
> $ ls build time.log $ cat build/time.log 0.00user 0.00system 0:00.00elapsed 0%CPU (0avgtext+0avgdata 1612maxresident)k 0inputs+0outputs (0major+73minor)pagefaults 0swaps 0.00user 0.00system 0:00.00elapsed 0%CPU (0avgtext+0avgdata 1616maxresident)k 0inputs+0outputs (0major+74minor)pagefaults 0swaps

closed 2023-02-15 by AugustusWillisWang