OpenXiangShan/XiangShan #1493: java.util.NoSuchElementException: key not found: ammonite.$file.rocket-chip.common

cisen, 2022-03-19:
```sh
cisen@cisen-virtual-machine:~/Desktop/develop/xiangshan/xiangshan-nanhu/xiangshan-nanhu-full-20220319$ make verilog
mkdir -p build
mill -i XiangShan.runMain top.TopMain -td build                      \
	--config DefaultConfig --full-stacktrace --output-file XSTop.v    \
	--infer-rw --repl-seq-mem -c:top.TopMain:-o:build/XSTop.v.conf \
	--gen-mem-verilog full --num-cores 1             \
	--disable-all --remove-assert --fpga-platform
Picked up _JAVA_OPTIONS: -XX:-UseGCOverheadLimit -Xms1024m -Xmx30g -Xss1024K
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at mill.main.client.IsolatedMillMainLoader.runMain(IsolatedMillMainLoader.java:58)
	at mill.main.client.MillClientMain.main(MillClientMain.java:67)
Caused by: java.util.NoSuchElementException: key not found: ammonite.$file.rocket-chip.common
	at scala.collection.MapOps.default(Map.scala:274)
	at scala.collection.MapOps.default$(Map.scala:273)
	at scala.collection.AbstractMap.default(Map.scala:405)
	at scala.collection.mutable.HashMap.apply(HashMap.scala:425)
	at mill.main.GraphUtils$.scriptNodeOf$1(GraphUtils.scala:16)
	at mill.main.GraphUtils$.$anonfun$linksToScriptNodeGraph$1(GraphUtils.scala:16)
	at scala.collection.immutable.List.map(List.scala:246)
	at scala.collection.immutable.List.map(List.scala:79)
	at mill.main.GraphUtils$.scriptNodeOf$1(GraphUtils.scala:16)
	at mill.main.GraphUtils$.$anonfun$linksToScriptNodeGraph$2(GraphUtils.scala:22)
	at scala.collection.Iterator$$anon$9.next(Iterator.scala:577)
	at scala.collection.mutable.Growable.addAll(Growable.scala:62)
	at scala.collection.mutable.Growable.addAll$(Growable.scala:57)
	at scala.collection.immutable.SetBuilderImpl.addAll(Set.scala:396)
	at scala.collection.immutable.Set$.from(Set.scala:353)
	at scala.collection.immutable.Set$.from(Set.scala:93)
	at scala.collection.IterableFactory$Delegate.from(Factory.scala:288)
	at scala.collection.IterableOps.map(Iterable.scala:675)
	at scala.collection.IterableOps.map$(Iterable.scala:675)
	at scala.collection.AbstractIterable.map(Iterable.scala:926)
	at mill.main.GraphUtils$.linksToScriptNodeGraph(GraphUtils.scala:22)
	at mill.main.RunScript$.importTree(RunScript.scala:139)
	at mill.main.RunScript$.$anonfun$runScript$6(RunScript.scala:81)
	at ammonite.util.Res$Success.map(Res.scala:67)
	at mill.main.RunScript$.runScript(RunScript.scala:71)
	at mill.main.MainRunner.$anonfun$runScript$1(MainRunner.scala:119)
	at mill.main.MainRunner.watchLoop2(MainRunner.scala:67)
	at mill.main.MainRunner.runScript(MainRunner.scala:92)
	at mill.MillMain$.main0(MillMain.scala:311)
	at mill.MillMain$.main(MillMain.scala:52)
	at mill.MillMain.main(MillMain.scala)
	... 6 more
make: *** [Makefile:57: build/XSTop.v] Error 1
```

- ubuntu 20.4
- java11
- scala 2.11.12

最新的master分支代码，多次尝试`make init`，也尝试过自己组合所有文件夹，都不行。yanqihu分支能成功编译。这个报错有人知道什么问题吗？

ljwljwljwljw, 2022-03-19:
`Caused by: java.util.NoSuchElementException: key not found: ammonite.$file.rocket-chip.common` 

* 请检查一下make init命令是否执行成功了，查看`rocket-chip`目录中是否完整的clone了upstream的文件，`rocket-chip/build.sc`文件是否存在?
* 使用`mill -v`查看mill版本是否是0.9.8?

cisen, 2022-03-19:
> `Caused by: java.util.NoSuchElementException: key not found: ammonite.$file.rocket-chip.common`
> 
> * 请检查一下make init命令是否执行成功了，查看`rocket-chip`目录中是否完整的clone了upstream的文件，`rocket-chip/build.sc`文件是否存在?
> * 使用`mill -v`查看mill版本是否是0.9.8?

降低mill到0.9.8确实可以，出错的是mill0.10.1

closed 2022-03-19 by cisen