OpenXiangShan/XiangShan #1519: how to simulate the DRAM time and power?

hanm2019, 2022-04-01:
In the [Doc](https://xiangshan-doc.readthedocs.io/zh_CN/latest/guide/), I found the DRAMsim3 simulator `[DRAMsim3](https://github.com/OpenXiangShan/DRAMsim3)：Cycle-level 地模拟内存的行为，针对香山项目进行了配置`, but I can not find where use the simulator

poemonsense, 2022-04-01:
`export DRAMSIM3_HOME=/path/to/DRAMsim3`
`make emu WITH_DRAMSIM3=1`

poemonsense, 2022-04-04:
可以看 `difftest/src/test/csrc/common/ram.cpp`

hanm2019, 2022-04-05:
请问cosimulation.h在哪呢？在xiangshan和difftest中都没有找到这个文件

poemonsense, 2022-04-05:
可以看 https://github.com/OpenXiangShan/DRAMsim3/blob/co-simulation/src/cosimulation.h

hanm2019, 2022-04-05:
在dramsim3/cosimulation.h的callback函数中，我对resp的finish_time的计算公式:
```
resp->finish_time = resp->req_time + (get_clock_ticks() - resp->req_time) * CPU_FREQ_SCALE + padding;
```
有一些疑问，在我看来,因为运行频率不同，系统时间涉及到cpu侧和dram侧。

1. `resp->finish_time`是使用dram侧周期数进行度量

依据：在check_respoinse函数中，
```
    auto now = get_clock_ticks();
    if (resp->finish_time <= now) 
```
2. `resp->req_time`也是使用dram侧周期数进行度量

依据：在add_request函数中
```
req_list.push_back(new CoDRAMResponse(request, get_clock_ticks()));
```

3. `get_clock_ticks()`本身也返回的是`dram_clock`

4. 私有参数CPU_FREQ_SCALE是用于将dram侧时钟数换算到cpu侧的

5. 因此，finish_time的计算公式的含义就变成了下面的3个部分之和：
（1） 此请求产生时的周期数（dram侧）
（2）dramsim3触发回调函数时的周期数（dram侧）与产生请求时的周期数（dram侧）之差换算到cpu侧的周期数
（3） padding周期数

其中，(2)和(1)&(3)的度量单位是不一样的，为什么可以直接相加呢？以及为什么要做(2)中的周期换算呢？

源代码：
```
void ComplexCoDRAMsim3::callback(uint64_t addr, bool is_write) {
    // std::cout << "cycle " << std::dec << get_clock_ticks() << " callback "
    //           << "is_write " << std::dec << is_write << " addr " << std::hex << addr << std::endl;
    // search for the first matched request
    auto iter = req_list.begin();
    while (iter != req_list.end()) {
        auto resp = *iter;
        if (resp->req->address == addr && resp->req->is_write == is_write) {
            req_list.erase(iter);
            resp->finish_time = resp->req_time + (get_clock_ticks() - resp->req_time) * CPU_FREQ_SCALE + padding;
            auto &queue = (resp->req->is_write) ? resp_write_queue : resp_read_queue;
            queue.push(resp);
            return;
        }
        iter++;
    }
    std ::cout << "INTERNAL ERROR: Do not find matched request for this response "
               << "(0x" << std::hex << addr << ", " << is_write << ")." << std::endl;
    abort();
}
```

poemonsense, 2022-04-06:
关于4，是将dram侧的绝对时间按照相对比例换算到cpu侧。

其实我们现在的计算是不合理的，因为现在dram侧和cpu侧事实上用的是同一个时钟（cpu tick一次，dram tick一次），只是通过 `CPU_FREQ_SCALE` 来硬乘上了一个系数，比如DDR2400对应CPU1200的话就需要乘2。

以上这样的做法是不对的，正确的做法应该是按照 `CPU_FREQ_SCALE` 来控制两边时钟的tick，只有时钟会受到频率比例的影响。我们现在的做法其实是导致对应CPU频率下的DDR的带宽降低了，仅仅通过手动加比例来调整了延迟。

关于5这一点，就像刚才说的，其实2是在dram时钟下的，硬加了一个比例放大，并不是cpu的周期数。或者换个角度，从您的这个分析来说，也可以认为现在dram和cpu在一个频率，但我们为了延迟对齐真实环境，手动加了一个延迟，我们这个做法是错误的。

这个问题我们一直都知道，没来得及动（其实改起来很简单，只要改一下 `ram.cpp` 里面对 dram 的 tick 就可以），过段时间我们会整体通过 `difftest` 环境的更新来处理掉。如果感兴趣的话也欢迎您帮忙直接改改

hanm2019, 2022-04-06:
看到`Emulator::single_cycle()`的处理过程大概理解为什么要乘一个系数了，我再看一看`testchipip`中关于dramsim2的实现，看看他怎么处理多个周期的

closed 2023-10-21 by cebarobot