OpenXiangShan/XiangShan #1521: 请问南湖coremark跑分7.81对应哪个分支？

qqjinger, 2022-04-08:
请问包老师2021.12.6宣称的南湖coremark跑分7.81对应南湖哪个分支以及哪个版本？谢谢。

poemonsense, 2022-04-08:
由于您在提问时没有提供具体的内容来源，我们只能猜测您说的“包老师2021.12.6宣称的南湖coremark跑分7.81”指的是包老师在RISC-V Summit上的报告PPT。

我们查看了当时的PPT内容，(1) PPT标明了 `Updated Nov.27 2021`，所以当时报告的 7.81 分是在2021年11月27日或之前的版本上跑出来的。我们看了一下代码提交记录，建议您可以尝试一下 6535afbb7b0abd815534ffb2a744878ed6c34ad1 这个版本的代码，这是11月27日之前合并到 master 分支的最新一个提交；(2) PPT标明了我们的编译选项，`RV64GCB O2`；(3) PPT标明了我们的CoreMark运行次数，也就是 `Running CoreMark for 100 iterations`。

南湖版本在包老师当时的报告后已经有很多的修改，包含因时序优化导致一些feature被砍掉，所以现在的CoreMark分数可能会比当时会低一些。您可以先尝试一下，如果发现复现不了，欢迎提供相关的git信息和对应的运行log给我们分析。

ljwljwljwljw, 2022-04-08:
编译Coremark时请注意开启b扩展: `-march=rv64gc_zba_zbb_zbc_zbs`

qqjinger, 2022-04-08:
谢谢，我克隆代码下来试试

poemonsense, 2022-04-11:
经过我们的再次确认：

基于 6535afbb7b0abd815534ffb2a744878ed6c34ad1 版本，100 轮 CoreMark RV64GCB O2 环境下报告的分数是 7.78，O3 环境下报告的分数是 8.73。

[image]
[image]

基于 nanhu 分支 509552226bb2a67da4f5b067d827f8a6e579e374 版本，100 轮 CoreMark RV64GCB O2 环境下报告的分数是 6.75，O3 环境下报告的分数是 7.90。

[image]
[image]

closed 2022-04-11 by poemonsense

qqjinger, 2022-04-11:
好的，非常感谢您的耐心细致回复