OpenXiangShan/XiangShan #1524: 关于编译RV64GCB 工具链失败的问题请教

qqjinger, 2022-04-11:
您好：

       我按照香山官方doc检出了对应版本的工具链。下图是香山官方docs\compiler\gnu_toolchain.md文件中检出工具链版本说明：
[image]
      下图则是我本地已检出版本的截图：
[image]
 ***
      但是，我按照官方的工具链编译方法编译时，出现了错误。
      下图是官方工具链编译方法：
[image]
       不过我按照上述编译步骤进行编译时，报告了错误：
[image]
[image]
        我查看了libgcc/config.log文件，发现错误信息详情如下：
[image]
[image]
        
       请问有谁遇到过这个问题吗？谢谢。

qqjinger, 2022-04-13:
请问编译rv64gcb工具链，对riscv toolchain mater分支的版本有没有要求？我看香山gnu_toolchain.md文件中没有说master版本。

ljwljwljwljw, 2022-04-13:
我在本地以前可以正常编译的版本也遇到了和您一样的问题，还在排查原因，之前我们使用的riscv toolchain版本是`b715e4f01b43efef487166f75d5d85d3c33fa7ef`.

qqjinger, 2022-04-16:
您好，我本地编译成功了，工具链版本如下：

- riscv toolchain版本回退到b715e4f01b43efef487166f75d5d85d3c33fa7ef；
- riscv-gcc 替换成软件所的版本 https://github.com/pz9115/riscv-gcc
  切换到 riscv-gcc-experimental 分支 e5ec1231dd03ec1e10807ab5677a600489fa017e
- riscv-binutils 替换成软件所的版本 https://github.com/pz9115/riscv-binutils-gdb
  切换到riscv-binutils-experimental 分支 520f5066004f7db320b39d09e9dd6d1a79202733

重要变化是，我根据编译错误信息，调整了工具链对应的指令集，修改后工具链编译命令如下：
```bash
cd riscv-gnu-toolchain
mkdir build
cd build
-  ../configure --prefix=PATH_TO_INSTALL --with-arch=rv64gc_zba_zbb_zbc_zbs
+  ../configure --prefix=PATH_TO_INSTALL --with-arch=rv64gc_zba_zbb_zbc
make linux -j
```

如此，编译没有问题。
不过有个问题，请问该riscv-gcc、riscv-binutils分支是不是不支持zbs扩展？

qqjinger, 2022-04-19:
对于gnu_toolchain.md文件中指定的gcc和binutils版本，通过查看`riscv-gcc/gcc/common/config/riscv/riscv-common.c`文件发现，该gcc支持1.0版zba_zbb_zbc_zbs扩展，但是`riscv-binutils/gas/config/tc-riscv.c`文件中指定只支持0.93版zba_zbb_zbc扩展，这应该就是按照gnu_toolchain.md文件中检出分支编译带zbs扩展出错的原因。

---
本地尝试了，采用如下版本，能成功编译出带1.0版zba_zbb_zbc_zbs扩展指令：

toolchain分支：master https://github.com/riscv/riscv-gnu-toolchain commit: b715e4f01b43efef487166f75d5d85d3c33fa7ef

gcc分支：riscv-gcc-experimental https://github.com/pz9115/riscv-gcc.git commit: https://github.com/pz9115/riscv-gcc/commit/ba72b9fb037a063f10cbaece3d73f56d48e11fcc Merge pull request https://github.com/pz9115/riscv-gcc/pull/11 from WuSiYu/riscv-gcc-experiment

binutils分支：riscv-binutils-experimental https://github.com/pz9115/riscv-binutils-gdb.git commit: 981aefbe0cb94e1fb0ee6aa03e8fc6421bf6ddf6

closed 2022-04-19 by qqjinger

ljwljwljwljw, 2022-04-20:
感谢您提的问题，[对应文档](https://xiangshan-doc.readthedocs.io/zh_CN/latest/compiler/gnu_toolchain/)已更新。