OpenXiangShan/XiangShan #1532: 有没有可能通过SimTop.v或者XSTop.v配以一个ram.v来进行最小化的RTL仿真？

trueif, 2022-04-18:
ram.v内容为microbench.bin中的代码，用RTL方式固化。
其它需要的xxHelper.v等，都砍掉。

poemonsense, 2022-04-18:
可以试试 https://github.com/OpenXiangShan/env-scripts/tree/main/vcs，是一个相对简单的仿真环境

poemonsense, 2022-04-19:
1 看您那边生成的rtl就可以，就是 `make verilog` 生成出来的那些 verilog 文件

2 上FPGA验证的话，需要使用FPGA上的DDR才行，靠FPGA综合得到的RAM很小。需要通过XSTop接口上的AXI连接FPGA板载的资源

trueif, 2022-04-19:
> 1 看您那边生成的rtl就可以，就是 `make verilog` 生成出来的那些 verilog 文件
> 
> 2 上FPGA验证的话，需要使用FPGA上的DDR才行，靠FPGA综合得到的RAM很小。需要通过XSTop接口上的AXI连接FPGA板载的资源

谢谢，我先试试看。
make emu产生的SimTop.v会和make verilog产生的XSTop.v有差别吗？

另，我想仿一个microbench.bin里面的内容，应该只要一个8192x64bit的内存就够吧？

poemonsense, 2022-04-19:
第一个问题，有差别。

第二个问题， @AugustusWillisWang 会确认一下。CoreMark肯定是够了，microbench不确定

trueif, 2022-04-19:
> 可以试试 [https://github.com/OpenXiangShan/env-scripts/tree/main/vcs，是一个相对简单的仿真环境](https://github.com/OpenXiangShan/env-scripts/tree/main/vcs%EF%BC%8C%E6%98%AF%E4%B8%80%E4%B8%AA%E7%9B%B8%E5%AF%B9%E7%AE%80%E5%8D%95%E7%9A%84%E4%BB%BF%E7%9C%9F%E7%8E%AF%E5%A2%83)

这个我试了，VCS可以仿真通过。
ram.c我改写成了ram.v。
目前还依赖了 flash.c, sdcard.c, uart,c 这几个c文件。我找找办法把它们阉割掉。
请问这几个C文件中，是否有对microbench.bin和linux.bin而言比较重要、不可去除或者必须找到Verilog等价替代品的？

poemonsense, 2022-04-19:
需要有uart，可以换个写法。flash和sdcard可以不要。

trueif, 2022-04-19:
> 需要有uart，可以换个写法。flash和sdcard可以不要。

在这俩bin的例程中，uart的功能主要是输出屏显结果吧？
如果我不要屏显结果，直接检查内存读取（比如microbench中的PASS字样、linux.bin中的Hello world字样），则也应该可以砍掉uart吧？

poemonsense, 2022-04-20:
可以试一试

trueif, 2022-04-21:
> 可以试一试

实测flash.c要改flash.v
uart.c可以砍掉，直接在tb里面$fwrite，或者注释掉。
sd.c可以直接砍。

closed 2023-02-15 by AugustusWillisWang