OpenXiangShan/XiangShan #1543: 在 Linux Kernel 跑 Spec2006 的问题

erdevonmatteo11, 2022-04-29:


ljh-kernel, 2022-11-12:
我想请问您，香山的RTL环境已经搭建好，请问要如何启内核进入系统，软件环境要怎么做？

wakafa1, 2023-04-19:
软件环境请参考香山文档前端开发环境一节：https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/xsenv/

wkxfudan, 2023-08-28:
> 我们采用您提到的的第一种方式，直接跑编译出来的benchmark elf文件，然后手动计算分数。 ref time / test time 是每个程序的分值，test time需要手动计时，ref time在源码的data/ref/reftime中，单位为秒。所有程序的几何平均就是最后的分值。

问个问题，对于一个case有多个负载的情况，如gcc，那test time是所有的负载加起来的总时间是吗

wakafa1, 2023-08-28:
> > 我们采用您提到的的第一种方式，直接跑编译出来的benchmark elf文件，然后手动计算分数。 ref time / test time 是每个程序的分值，test time需要手动计时，ref time在源码的data/ref/reftime中，单位为秒。所有程序的几何平均就是最后的分值。
> 
> 问个问题，对于一个case有多个负载的情况，如gcc，那test time是所有的负载加起来的总时间是吗

是的，加起来

closed 2023-10-21 by cebarobot

Ergou-ren, 2024-02-26:
你好，我有一个问题，是不是香山的spec2006的测试需要跑在linux中？还是说，我可以不启动linux内核，直接跑spec2006的程序？

wakafa1, 2024-02-26:
需要 Linux 提供的运行时环境

cebarobot, 2024-02-26:
> 你好，我有一个问题，是不是香山的spec2006的测试需要跑在linux中？还是说，我可以不启动linux内核，直接跑spec2006的程序？

请注意，您提出了一个新问题，请您新开一个 issue 提问。
Please note that you've raised a new question. Please open a new issue to ask it.