OpenXiangShan/XiangShan #1549: perf model?

cnjsdfcy, 2022-05-10:
Hi,

不知道Xiangshan团队内部是否有基于Gem5开发perf model？还是直接基于Chisel快速实现、验证各种features的性能trade-off?

谢谢~

Lingrui98, 2022-05-10:
参见 #1495 
这两代架构并没有基于Gem5开发性能模型，而是用RTL仿真进行设计空间探索，不过后续会有这样的规划

cnjsdfcy, 2022-05-10:
两个问题：
1. 是否因为在开发过程中还是觉得https://github.com/OpenXiangShan/XiangShan/issues/1495#issuecomment-1077284804 这样的性能评估不够快捷？便利？
2. 原本我有一点体会是Chisel语言即兼顾了silicon RTL的生成，又兼顾了快速设计空间的探索，目前看来并不是这样，您感受如何？

sequencer, 2022-05-10:
本质是perfmodel与rtl的对齐和rtl simu效率问题的tradeoff

Lingrui98, 2022-05-10:
初期受限于人力和项目进度要求，我们选择直接用chisel开发RTL，在它上面进行评估和一定的设计空间探索，我们也通过各种手段加速了RTL开发和探索的过程。
在这个前提下，模拟器评估方法下修改代码和运行的效率仍会较高一些，但模拟器需要一个开发过程，这是一个门槛，当有条件跨过这道门槛的时候，就可能会去选择模拟器评估。

cnjsdfcy, 2022-05-10:
> 本质是perfmodel与rtl的对齐和rtl simu效率问题的tradeoff

Sequencer大佬一语道出本质。
是的，perf model会存在一个很大的effort是perf correlation，往往在工程中需要花费较多的精力投入。

如果对perf modeler提升一点能力要求（更细节的电路建模），直接用Chisel构建RTL，是不是可以避免掉对齐的开销？

另一点，关于RTL simu效率，Xiangshan团队的Betapoint，从分享出的信息看，貌似比Gem5 + simpoint速度快不少，不知道这点是不是可以进一步增强直接用Chisel进行架构探索的可能性？

closed 2022-05-12 by cnjsdfcy