OpenXiangShan/XiangShan #1606: 请问香山的FE在后级发生Flush之后的具体行为是怎么样的

Linz1zz, 2022-07-01:
从文档中得到当后级执行单元对分支重定向的请求会发送到FTQ，那FTQ是可以立即开始向IFU发起取指请求吗？还是说要等BPU对当前预测块的预测结果？

Lingrui98, 2022-07-01:
Translation of the question：
I read your document and found that branch misprediction redirect from execution units would be sent to FTQ. Can FTQ send fetch request to IFU immediately in that case? Or should it wait for BPU to produce a prediction of the block starting with the new target?

目前的做法是等BPU对新预测块的预测结果。如果直接向IFU发起取指请求的话，会涉及到预测块内部分flush的情况，这种情况会引入比较复杂的控制逻辑，目前并没有实现。

Currently FTQ should wait for the new prediction from BPU. If FTQ is allowed to send fetch request by itself, there could be a problem of partial flushing a fetch(prediction) block, which would introduce complex control logic. So such mechanism is not implemented yet.

RedClothesBoy, 2022-07-12:
感谢你的回复，另外还有一个点，我发现FE访问icache时，idata读位宽分别为64 Byte和128 Byte(指令block跨cacheline)，感觉这样的设计对于功耗和Timing均不友好，为什么此处不采用32 Byte和64 Byte访问位宽呢

cebarobot, 2024-09-14:
We see that you haven't responded to this issue for a long time. If you have any other questions, please feel free to open the issue again or file a new one.

我们看到您已经很久没有回复本 issue。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

closed 2024-09-14 by cebarobot