OpenXiangShan/XiangShan #1609: 请问如何控制xstop通过memory口及时写外部ddr？

qqjinger, 2022-07-05:
您好：
       目前我可以通过xstop的memory从外部读取指令，但是通过memory口直接写外部ddr时，发现数据并没有反应在xstop的memory端口上。
      怀疑xstop将数据缓存在L3 cache，一直没有将数据提交到外部ddr。
      请问我该如何通过memory口及时有效的对外部ddr设备进行写操作？谢谢。

wakafa1, 2022-07-06:
要实现这种需求，一般是通过 Cache 指令把缓存中的数据刷到 DDR 中。RISC-V 有一个 CMO Task Group 正在制定规范，可以参考：https://github.com/riscv/riscv-CMOs

wakafa1, 2022-07-06:
要实现这种需求，一般是通过 Cache 指令把缓存中的数据刷到 DDR 中。RISC-V 有一个 CMO Task Group 正在制定规范，可以参考：https://github.com/riscv/riscv-CMOs
由于该规范尚未冻结，香山没有实现一套完整的缓存控制指令。我们在内部实现了一个简易的 L3 缓存操作控制器，通过 MMIO 来控制，目前没有文档，有一个简单的驱动可供参考 https://github.com/OpenXiangShan/nexus-am/blob/master/tests/cacheoptest/llc/src/huancunop.c

qqjinger, 2022-07-12:
您好，请问huancunop.c](https://github.com/OpenXiangShan/nexus-am/blob/master/tests/cacheoptest/llc/src/huancunop.c)[huancunop.c](https://github.com/OpenXiangShan/nexus-am/blob/master/tests/cacheoptest/llc/src/huancunop.c)的test3 flush cache中test1-3的invalid cache/clean cache/flush cache分别表示什么含义？
        目前我用20220124default emu测试发现，test1 invalid cache失败，test2 clean cache和test3 flush cache成功。
       打开保存的波形，发现flush cache并没有通过mem端口发现往芯片外部写数据。

qqjinger, 2022-07-12:
您好，请问huancunop.c](https://github.com/OpenXiangShan/nexus-am/blob/master/tests/cacheoptest/llc/src/huancunop.c)[huancunop.c](https://github.com/OpenXiangShan/nexus-am/blob/master/tests/cacheoptest/llc/src/huancunop.c) 中test1-3的invalid cache/clean cache/flush cache分别表示什么含义？
        目前我用20220124default emu测试发现，test1 invalid cache失败，test2 clean cache和test3 flush cache成功。
       打开保存的波形，发现flush cache并没有通过mem端口发现往芯片外部写数据。

wakafa1, 2022-07-12:
invalidate/clean/flush 的定义可以参考 [riscv-CMOs](https://github.com/riscv/riscv-CMOs)  的暂定规范；该测试会影响缓存一致性，因此需要在仿真时关闭 difftest（--no-diff 选项）。另外，请使用最新版本的香山进行测试，我们几个月前对缓存操作控制器有过修改

user-stu-l, 2022-12-02:
您好，我也遇到类似问题，在关闭diff的情况下进行操作，test1 invalid cache报以下错误，其它两个test可以通过，请问这种情况如何处理呢：
HuanCun l3 size is set to 6144 KB, nr_way is set to 6, nr_bank is set to 4, nr_set is 4096, set_len is 12
addr 0x80001940 tag 0x2000 set 0x65
------------------------------------------------------------
huancun op invalid done
data 1
CMO_INV failed: data right 1 wrong 1test failed.
Core 0: HIT BAD TRAP at pc = 0x800002ca
total guest instructions = 11,821
instrCnt = 11,821, cycleCnt = 40,696, IPC = 0.290471

AugustusWillisWang, 2023-02-15:
> 您好，我也遇到类似问题，在关闭diff的情况下进行操作，test1 invalid cache报以下错误，其它两个test可以通过，请问这种情况如何处理呢：
> 
> huancun op invalid done data 1 CMO_INV failed: data right 1 wrong 1test failed. Core 0: HIT BAD TRAP at pc = 0x800002ca total guest instructions = 11,821 instrCnt = 11,821, cycleCnt = 40,696, IPC = 0.290471

缓存操作控制器的问题近期已经修复。

closed 2023-05-22 by wakafa1