OpenXiangShan/XiangShan #1629: 请问香山fpga项目运行linux该如何交互？emu打印机制？

qqjinger, 2022-07-08:
您好：
        编译香山后，在XSTop.dts中看到有串口地址定义：
	L15: soc {
        		L1: serial@40600000 {
			compatible = "xilinx,uartlite";
			reg = <0x0 0x40600000 0x0 0x10>;
		};
       但是编译生成的xstop顶层并没有看到串口IO端口，只有一个jtag口。
       有两个问题：
       1）如果使用生成的XSTop.v，打包成FPGA工程，以运行可交互的linux，是否需要自己在FPGA端自己创建xilinx uart ip？
       2）emu运行testcase.bin文件的打印，是不是不用通过串口，直接就输出在屏幕上了？如果emu不用串口打印，那么emu是通过哪种机制对c语言中print这些函数打印在屏幕上？
       谢谢。

wakafa1, 2022-07-09:
（1）是的，需要在 FPGA 上配一个 UART，根据 FPGA 上 UART IP 的地址空间和类型修改串口打印驱动
（2）emu 运行时是通过一个仿真的虚拟串口进行输出的，可以参考https://github.com/OpenXiangShan/difftest/blob/0e1242e150d5ab3069075820107709b3e29dc43f/src/test/csrc/verilator/emu.cpp#L319

qqjinger, 2022-07-11:
>非常感谢您的回复

closed 2023-02-15 by AugustusWillisWang