OpenXiangShan/XiangShan #1671: 关于香山验证请教

qqjinger, 2022-07-21:
您好：
        注意到目前香山主要是通过运行一些特定的测试用例，比如coremark\dhrystone\spec\linux等应用程序来进行系统仿真，通过difftest和nemu进行差分测试实时判断结果是否正确。
       这种特定测试用例的方法，可能不足以找出香山所有的bug。
       有几个问题想请教下：
      1、请问是否有对内核模块，做全面的UVM模块验证？
      2、emu运行测试用例，是否有方式统计覆盖率？
      3、是否可以用诸如riscv-torture 这样的随机指令生成平台接入香山，持续生成随机指令到emu执行并进行差分测试，直到处理器报错差分测试未通过再结束仿真？然后根据保存的报错前波形，定位响应bug。感觉这样验证可能更为充分。

poemonsense, 2022-07-21:
基于应用程序来仿真，确实不足以找出香山所有的bug。

关于第一个问题，我们正在和合作伙伴一起搭建模块级的验证框架。UVM是方法学和验证工具，使用UVM不代表验证质量就有保证。

关于第二个问题，verilator和vcs都支持覆盖率的统计。

关于第三个问题，可以使用riscv-torture, riscv-dv, force-riscv等框架生成随机指令，灌给香山执行。关于验证质量和验证充分的问题，您可以基于这几个框架在香山上试验运行一下，看看是否能发现更多的问题。riscv-torture在香山项目下有fork，可以直接生成；riscv-dv也可以基于cfg来生成汇编，基于香山的链接脚本生成测试。我们都跑过，您也可以试着跑一下

poemonsense, 2022-07-21:
非常感谢您的建议，也欢迎提供具体的实现思路，帮助我们提高验证质量

qqjinger, 2022-07-21:
前面第三个问题您提到香山有跑过各种随机指令生成平台生成的指令，请问香山运行这些百亿级别随机指令后，有收集覆盖率吗？如果我想使用emu收集覆盖率，是否需要自己实现这部分功能？

poemonsense, 2022-07-21:
我们目前还没有跑过百亿级别的随机指令，riscv-dv生成数量大的测试之后，.S的大小会膨胀，编译不了。应该是只能编译大量的短片段，然后并行跑。另外百亿级别的测试仿真会非常慢，我们一般大概会并行跑<1B级别长度的测试。超过10B的基本上是个别长时间挂着跑了，迭代也挺慢的，我们一般只有在长时间回归的时候才跑这么长的

emu在make的时候有一个选项EMU_COVERAGE，可以参考verilator.mk。VCS的话也是类似，需要您那边单独实现一下

qqjinger, 2022-07-21:
非常感谢您的回复，学到了，我试试用emu收集覆盖率。
另外，想请教下，emu仿真和rtl仿真，行为是否能保证完全一致？毕竟最终流片的是rtl代码。

poemonsense, 2022-07-21:
香山代码生成的emu就是Verilog RTL仿真，和流片的RTL是一致的。我们可能命名上不是很合适，emu指的不是emulator，就是普通的rtl仿真程序

qqjinger, 2022-07-28:
请问香山运行1B随机指令时，如果和nemu差分测试出错，那么如何定位问题？如果把emu运行的所有1B随机指令波形全部保存在VCD波形中，那么波形会特别大，感觉保存全时间段的信号到波形不太现实。

poemonsense, 2022-07-28:
有两个方法。

第一个，`--dump-wave -b X -e Y`只会在X和Y周期之间打波形

第二个，`--enable-fork`会打开LightSSS工具，自动恢复最后一小段时间内的波形

Jishuaishuai123, 2022-09-20:
> 基于应用程序来仿真，确实不足以找出香山所有的bug。
>
> 关于第一个问题，我们正在和合作伙伴一起搭建模块级的验证框架。UVM是方法学和验证工具，使用UVM不代表验证质量就有保证。
>
> 关于第二个问题，verilator和vcs都支持覆盖率的统计。
>
> 关于第三个问题，可以使用riscv-torture, riscv-dv, [...]

您好：我使用riscv-torture工具，在生成几十万条指令的时候就会出现：java.lang.OutOfMemoryError: Java heap space，后面通过配置JVM的参数来解决（ export JAVA_TOOL_OPTIONS(_JAVA_OPTIONS)=" -Xmx8G -Xms8G -Xmn3G -Xss512k -XX:NewRatio=4 -XX:SurvivorRatio=4 -XX:MaxPermSize=128k -XX:MaxTenuringThreshold=0 -XX:MaxTenuringThreshold=16 -XX:+UseConcMarkSweepGC"），百万条指令生成是有效的，目前在生成千万级别的时候需要不断调大-Xmx的内存，如果配置-Xmx的内存过大程序运行一段时间就会被杀死(make: *** [Makefile:44: igentest] 已杀死)，过小就一直报warn([warn] In the last 70 seconds, 62.819 (90.7%) were spent in GC. [Heap: 21.66GB free of 30.00GB, max 30.00GB] Consider increasing the JVM heap using `-Xmx` or try a different collector, e.g. `-XX:+UseG1GC`, for better performance.)
对此想请教您：
①请问你们是如何解决java.lang.OutOfMemoryError: Java heap space这个问题的；
②我将sequence配置为100w时生成.S文件需要22h，请问您这边生成1亿、10亿条指令分别用了多长时间；

poemonsense, 2022-09-20:
我们没有使用过torture来生成M或者B级别的指令长度

SeddonShen, 2022-12-19:
麻烦请教下，XiangShan针对特权指令集部分的验证的方法是怎么样的，是通过difftest吗，跑的测例是什么样子的呢？

qqjinger, 2023-02-11:
有3个问题想请教下：

1、请问能分享下香山发现bug速度快的方法吗？
分解为如下几个问题：
1.1、发现bug方式或运行哪些测试用例？
注意到香山发现的bug涉及各个模块，这些仅仅系统验证通过运行nexus-am下的所有apps，就能发现这些bug吗？
有一些cache、总线、mmu、pmp、pma等方面的设计错误，感觉需要运行运行复杂测试用例，且在一些极端情况下才可能出现bug，nexus-amp/apps下的这些用例感觉不一定能触发这些极端场景。
1.2、cache是否有专门的验证对数据正确性进行保证？
另外，不知道difftest是否能对各种情况下cache的一致性进行对比？或者有香山有其它方式专门验证cache的数据一致性？
例如：复杂场景下，可能cache的cacheline替换实现错误，导致错误位置的数据被替换，这个是否可以有方式快速发现类似的错误？

2、请问香山如何解决运行复杂测试用例debug难的问题？
我在本地尝试过三个级别指令数测试用例的运行：
**2.1、百万级指令数测试用例：迭代速度快，但场景不够复杂**
defaultConfig emu运行2次迭代的coremark（执行指令数约60万条），开启与NEMU差分测试，只需几十分钟就可出结果，并导出波形，这个级别指令数测试用例，出现问题定位很方便，但是测试用例复杂度较低，无法对内核极端或复杂场景进行验证。
2.2、1亿左右指令数测试用例：迭代速度较慢
defaultConfig emu运行1亿执行指令数的linux，我本地大约需要3天。如果差分测试出问题，那么通过--enable-fork是可以导出最后出错一段的波形，也能定位问题。
2.3、1万亿左右指令数测试用例：emu运行速度太慢，完全没法进行仿真定位问题
defaultConfig emu运行1万亿执行指令数的linux+spec的一个测试用例，大于需要3*1万=3万天。使用emu运行spec的一个完整测试用例，完全不现实。注意到香山使用checkpoint方式，采样了spec部分程序段，然后到emu上运行，然后得到了估算的spec分数。但有个问题，怎么保证内核（仿真或流片后）运行1万亿指令数的spec时，不会报错或者内核运行正常？毕竟没有完整跑过这个万亿级别的程序。

3、请问emu和NEMU同时运行指令app，通过difftest差分测试，是否意味着内核mem或mmio口的输出也没问题？
注意到difftest是定义了一些差分测试事件，cache相关的有sbuffer、store、load、actomic、ptw、refill。如果在某个测试用例中，store到dcache的数据是正确的，但是dcache->L3cache->mem_IO->核外ddr的内核输出数据出错了，是否能通过difftest来发现这样的错误？

poemonsense, 2023-10-21:
1.1 特定模块需要定向测试，或模块测试。difftest只解决isa可见错误的识别问题，不解决测试生成的问题

1.2 可以关注difftest中的refill test和tl-test项目https://github.com/OpenXiangShan/tl-test

2.3 目前正在做仿真加速器（整理代码中，很快合并），和FPGA环境上的difftest，以及优化后续的调试问题

3. difftest只能检查指令集层面可见的结果不一致，如果错误行为没有传播到指令集可见的层面，difftest无法发现。此时问题会回归到问题1中的问题