OpenXiangShan/XiangShan #1770: Make emu bug

Cranexiao95, 2022-09-05:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior. Please include the commit hash and your modification as well.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (optional, if necessary):**
 - OS: [e.g. Ubuntu 20.04 LTS]
 - Compiler: [e.g. GCC 9.4.0]

**Additional context**
Add any other context about the problem here.

poemonsense, 2022-09-05:
是不是配置verilator的时候安装了ccache？

Cranexiao95, 2022-09-06:
> 是不是配置verilator的时候安装了ccache？

谢谢回答，发现是我的verilator版本旧导致的

closed 2022-09-06 by Cranexiao95