OpenXiangShan/XiangShan #1796: make verilog erro

qiuchaochaoqaq, 2022-10-10:
make verilog erro,The error is reported as follows：
1 target failed
XiangShan.runMain subprocess failed
make:***[Makefile:57: build/XSTop.v]

wakafa1, 2022-10-10:
It is most likely caused by insufficient memory, please refer to #1254

qiuchaochaoqaq, 2022-10-10:
> It is most likely caused by insufficient memory, please refer to #1254

I have changed 64 to 15. I still report this error

qiuchaochaoqaq, 2022-10-10:
@wakafa1

wakafa1, 2022-10-10:
How much memory available do you have?

wakafa1, 2022-10-10:
Sorry for misunderstanding. I mean your DDR memory instead of disk space

qiuchaochaoqaq, 2022-10-10:
Do you mean the memory allocated to the virtual machine? My laptop has only 8G of storage, and I allocate 6G to the virtual machine @wakafa1

wakafa1, 2022-10-10:
For compiling Xiangshan, the memory capacity is too small. We recommend 32GB for minimalconfig while 64GB for defaultconfig.
You can continuously reduce memory size in build.sc or enlarge your swap, but it may take much longer to compile.

qiuchaochaoqaq, 2022-10-10:
ok.thank you  very much @wakafa1

wakafa1, 2022-10-10:
My pleasure. May I close this issue?

qiuchaochaoqaq, 2022-10-10:
sure @wakafa1

closed 2022-10-10 by wakafa1