OpenXiangShan/XiangShan #1807: Make verilog error

zhonglinghao, 2022-10-21:
我的是Ubuntu20.04LTS，按照开发环境部署配置，init以后make erilog出现以下报错问题，麻烦各位大佬指点谢谢
# There is insufficient memory for the Java Runtime Environment to continue.
# Native memory allocation (mmap) failed to map 19428016128 bytes for committing reserved memory.
# An error report file with more information is saved as:
# /home/zlh/桌面/xs-env/XiangShan/hs_err_pid18523.log
1 targets failed
XiangShan.runMain subprocess failed
make: *** [Makefile:57：build/XSTop.v] 错误 1

wakafa1, 2022-10-21:
报错信息已经写在你发的信息中的："There is insufficient memory for the Java Runtime Environment to continue."
可以参考 #1796

zhonglinghao, 2022-10-22:
> 报错信息已经写在你发的信息中的："There is insufficient memory for the Java Runtime Environment to continue." 可以参考 #1796
您好我也尝试去build.sc修改到32G，16G，8G和6G都是会出现报错情况。请问应该如何解决？

wakafa1, 2022-10-22:
你的内存容量是多大？在编译的时候有观察到占用内存接近上限的情况嘛？

zhonglinghao, 2022-10-22:
[image]
您好我的内存容量是8G，编译得时候是双系统编译只运行该程序。。

good-circle, 2022-10-22:
目前对于编译完整的香山核，我们推荐内存至少为 64GB；对于编译 MinimalConfig 选项的香山核，我们推荐内存至少为 32GB。对于不满足以上内存的情况，可以考虑减小 -Xmx 的值，或者考虑改变 swap 区的大小（参见 [Troubleshooting 指南](https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/troubleshoot-zh/)）。但香山本身是一个很庞大的项目，-Xmx 过小并不足以支撑编译香山。

之前你提到 ：
> 您好我也尝试去build.sc修改到32G，16G，8G和6G都是会出现报错情况。

但报错的具体内容应该并不相同，-Xmx 足够大时机器的内存容量不够；将 -Xmx 调小后虽然满足机器的内存容量，但并不足以支撑编译香山。如果想要编译完整的香山核，需要至少满足 -Xmx 为 32G。

针对目前的情况，我更推荐通过 `CONFIG=MinimalConfig` 选项编译 MinimalConfig 的香山。我使用的环境是 Ubuntu 20.04 的虚拟机，CPU 为 Intel Core i7-12700H，经过验证，MinimalConfig 香山只需要至少满足 -Xmx 为 12G。有关 MinimalConfig 的说明，请参考 [香山前端开发环境](https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/xsenv/)。

zhonglinghao, 2022-11-24:
您好我在make verilog时没有报错也生成了XSTop.v文件但是我在生成香山核的仿真程序时却出现了报错，这是运行内存不足吗？下图是我的配置情况。谢谢各位大佬指点
[image]

make emu CONFIG=MinimalConfig SIM_ARGS=--disable-log EMU_TRACE=1 -j32
make -C ./difftest emu SIM_TOP=SimTop DESIGN_DIR=/home/lu/桌面/xs-env/XiangShan NUM_CORES=1
make[1]: 进入目录“/home/lu/桌面/xs-env/XiangShan/difftest”
make -C /home/lu/桌面/xs-env/XiangShan sim-verilog
make[2]: 进入目录“/home/lu/桌面/xs-env/XiangShan”
mkdir -p build
Thu, 24 Nov 2022 23:44:07 +0800
time -a -o ./build/time.log mill -i XiangShan.test.runMain top.SimTop -td build  \
	--config MinimalConfig                                            \
	--infer-rw --repl-seq-mem -c:top.SimTop:-o:build/SimTop.v.conf --gen-mem-verilog full                                               \
	--num-cores 1                                      \
	--disable-log --enable-difftest 
[236/236] XiangShan.test.runMain 
Exception in thread "main" firrtl.options.OptionsException: Unknown option --disable-log
	at firrtl.options.ExceptOnError.reportError(OptionParser.scala:25)
	at firrtl.options.ExceptOnError.reportError$(OptionParser.scala:25)
	at firrtl.options.Shell$$anon$1.reportError(Shell.scala:19)
	at scopt.OptionParser.handleError$1(options.scala:453)
	at scopt.OptionParser.parse(options.scala:513)
	at firrtl.options.Shell$$anon$1.firrtl$options$DuplicateHandling$$super$parse(Shell.scala:19)
	at firrtl.options.DuplicateHandling.parse(OptionParser.scala:51)
	at firrtl.options.DuplicateHandling.parse$(OptionParser.scala:32)
	at firrtl.options.Shell$$anon$1.parse(Shell.scala:19)
	at firrtl.options.Shell.parse(Shell.scala:58)
	at firrtl.options.Stage.execute(Stage.scala:58)
	at top.Generator$.execute(Generator.scala:76)
	at top.SimTop$.main(SimTop.scala:115)
	at top.SimTop.main(SimTop.scala)
1 targets failed
XiangShan.test.runMain subprocess failed
make[2]: *** [Makefile:100：build/SimTop.v] 错误 1
make[2]: 离开目录“/home/lu/桌面/xs-env/XiangShan”
make[1]: *** [Makefile:34：/home/lu/桌面/xs-env/XiangShan/build/SimTop.v] 错误 2
make[1]: 离开目录“/home/lu/桌面/xs-env/XiangShan/difftest”
make: *** [Makefile:139：emu] 错误 2

Lemover, 2022-11-24:
编译输出中提到：**Unknown option --disable-log**
编译参数中的 --disable-log 已经被去掉了，您可以去掉这个参数，重新尝试一下

zhonglinghao, 2022-11-24:
> 编译输出中提到：**Unknown option --disable-log** 编译参数中的 --disable-log 已经被去掉了，您可以去掉这个参数，重新尝试一下

您好，请问下这个参数是在哪个文件目录下的？是XiangShan/src/main/scala/ 里面的哪个呢？谢谢您。

Lemover, 2022-11-24:
--disable-log 参数已经被去掉了

zhonglinghao, 2022-11-24:
> --disable-log 参数已经被去掉了

那我是需要再次运行这个code吗？我已经再次运行还是报告这个错误。。。谢谢您

Lemover, 2022-11-24:
再次运行时，**输入的命令**去掉这个参数

zhonglinghao, 2022-11-24:
> 再次运行时，**输入的命令**去掉这个参数

okk已经解决非常感谢您

zhonglinghao, 2022-11-24:
[image]
您好各位大佬我想请问下我仿真环境安装好了在运行wordload的时候出现以上报错。

Lemover, 2022-11-24:
可以使用xiangshan目录下的 ready-to-run 下，已经准备好的 riscv64-nemu-interpreter-so 文件。
图中的错误，是没有设置环境变量 NEMU_HOME。

zhonglinghao, 2022-11-24:
> 可以使用xiangshan目录下的 ready-to-run 下，已经准备好的 riscv64-nemu-interpreter-so 文件。 图中的错误，是没有设置环境变量 NEMU_HOME。

您好如何设置环境变量NEMU_HOME?
我尝试运行riscv64-nemu-interpreter-so文件出现错误，请问这是什么情况？谢谢您。
[image]

Lemover, 2022-11-24:
使用方式：./build/emu -i ready-to-run/linux.bin --diff  ready-to-run/riscv64-nemu-interpreter-so

zhonglinghao, 2022-11-25:
> 使用方式：./build/emu -i ready-to-run/linux.bin --diff ready-to-run/riscv64-nemu-interpreter-so

您好大佬请问下我用这个./build/emu -i ready-to-run/linux.bin --dump-wave -b 10000 -e 11000 --diff ready-to-run/riscv64-nemu-interpreter-so 命令生成的波形文件在哪个目录下？谢谢您。
[image]

Lemover, 2022-11-25:
如果编译香山时，没有加上参数 EMU_TRACE=1，运行emu程序时，也不会打印波形。具体请参考香山开发手册 https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/xsenv/#_5
生成的波形在 ./build/***.vcd

zhonglinghao, 2022-11-25:
[image]
请问我应该如果和设置这个NOOP_HOME?谢谢您

Lemover, 2022-11-25:
参考香山使用手册：https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/xsenv/

zhonglinghao, 2022-11-25:
> 参考香山使用手册：https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/xsenv/

您好大佬，那我是要在/xs-env/NEMU目录下运行./build/riscv64-nemu-interpreter -b $NOOP_HOME/ready-to-run/linux.bin
把$NOOP_HOME替换成xs-env/NEMU然后在运行make emu CONFIG=MinimalConfig EMU_TRACE=1 -j32 吗？
谢谢您。

closed 2023-02-11 by Lemover