OpenXiangShan/XiangShan #1837: LRSC匹配失败问题

kl123s, 2022-11-25:
**Describe the bug**
出现断言报错：
![image](https://user-images.githubusercontent.com/119037558/203945188-39287665-66f7-4509-85d1-33cdd21d6021.png
“L1DCache failed too many SCs in a row”
  assert(debug_sc_fail_cnt < 100.U, "L1DCache failed too many SCs in a row")
这里为什么要设置允许sc存储失败不超过100次。

AugustusWillisWang, 2022-11-29:
https://github.com/OpenXiangShan/XiangShan/blob/5da19fb3f5e30e8e3654dcd8ba1fefc3f257bb3a/src/main/scala/xiangshan/cache/dcache/mainpipe/MainPipe.scala#L583
https://github.com/OpenXiangShan/XiangShan/blob/5da19fb3f5e30e8e3654dcd8ba1fefc3f257bb3a/src/main/scala/xiangshan/cache/dcache/mainpipe/MainPipe.scala#L602

master 上相关的 assertion 已更新

kl123s, 2022-12-01:
> https://github.com/OpenXiangShan/XiangShan/blob/5da19fb3f5e30e8e3654dcd8ba1fefc3f257bb3a/src/main/scala/xiangshan/cache/dcache/mainpipe/MainPipe.scala#L583
> 
> 
> https://github.com/OpenXiangShan/XiangShan/blob/5da19fb3f5e30e8e3654dcd8ba1fefc3f257bb3a/src/main/scala/xiangshan/cache/dcache/mainpipe/MainPipe.scala#L602
> 
> master 上相关的 assertion 已更新.原有的断言过于严格, 可以删掉

请问一下，原子指令的lr和sc不应该是一一对应的吗，为什么可以允许出现过多的sc指令。

AugustusWillisWang, 2023-02-15:
这里的代码仅仅用于辅助调试。在 SC 指令多次连续执行失败的时候发出警告。

closed 2023-10-21 by Tang-Haojin