OpenXiangShan/XiangShan #1870: Dockerfile is prefered for building developing environment

backyes, 2023-01-03:
https://xiangshan-doc.readthedocs.io/zh_CN/latest/tools/xsenv/ details on building developing environment and it will abort within docker environment by default. 

Docker is perfered for releasing developing and runtime envrionments. I think Xiangshan's frontend development tools can work in docker containers entirely. If community agree with it, I would like to contribute the Dockerfile style release environment for https://github.com/OpenXiangShan/xs-env  repo.

wakafa1, 2023-01-11:
Yes, we think docker is an important frontend development environment for beginners. We will appreciate it if you participate in the contribution.

backyes, 2023-01-15:
Ok, I will do that.

closed 2025-09-10 by Yan-Muzi