OpenXiangShan/XiangShan #1884: 请教关于异常委托的问题

heiyuen1999, 2023-02-01:
在测试S模式异常委托功能时，发现当medeleg委托权限打开后，在S模式下触发的加载访问错误异常（异常编号5）无法委托至S模式，仍是交由M模式处理。

[image]
此外，还有如下异常无法委托到S模式：存储器访问错误异常（7），指令页错误异常（编号12），加载页错误异常（编号13），存储器错误异常（编号15）
留意到他们都是非精确异步异常，是否因为此原因无法进行异常委托？

Lemover, 2023-02-01:
您可以提供一下 CSR 模块内部的对应信号的波形么

heiyuen1999, 2023-02-01:
> 

可以，请问您的邮箱是？

heiyuen1999, 2023-02-01:
> 您可以提供一下 CSR 模块内部的对应信号的波形么

可以 请问您的邮箱是？

Lemover, 2023-02-01:
> > 您可以提供一下 CSR 模块内部的对应信号的波形么
> 
> 可以 请问您的邮箱是？

您可以以类似上面的形式，直接发图片。

这些类型的例外，在系统中（如Linux系统）中，是比较常见的例外，所以我更倾向于测试用例编写有问题。如果您方便分享测试用例的源码，我们能更快定位问题。

heiyuen1999, 2023-02-01:
> > > 您可以提供一下 CSR 模块内部的对应信号的波形么
> > 
> > 
> > 可以 请问您的邮箱是？
> 
> 您可以以类似上面的形式，直接发图片。
> 
> 这些类型的例外，在系统中（如Linux系统）中，是比较常见的例外，所以我更倾向于测试用例编写有问题。如果您方便分享测试用例的源码，我们能更快定位问题。

> > > 您可以提供一下 CSR 模块内部的对应信号的波形么
> > 
> > 
> > 可以 请问您的邮箱是？
> 
> 您可以以类似上面的形式，直接发图片。
> 
> 这些类型的例外，在系统中（如Linux系统）中，是比较常见的例外，所以我更倾向于测试用例编写有问题。如果您方便分享测试用例的源码，我们能更快定位问题。
以下是测试用例
[image]

[image]

heiyuen1999, 2023-02-02:
> > > 您可以提供一下 CSR 模块内部的对应信号的波形么
> > 
> > 
> > 可以 请问您的邮箱是？
> 
> 您可以以类似上面的形式，直接发图片。
> 
> 这些类型的例外，在系统中（如Linux系统）中，是比较常见的例外，所以我更倾向于测试用例编写有问题。如果您方便分享测试用例的源码，我们能更快定位问题。
[image]

[image]
以上是测试用例。置SUM=0，通过访问一个U=1的页表触发load access fault

Lemover, 2023-02-02:
您可以贴一下 CSR 模块内部的对应信号么，包括先不限于 priviledgeMode, m/s -cause, m/s status, m/s tval, csrio.exception.bits.uop.cf.exceptionVec, csrio.exception.valid.

Lemover, 2023-02-02:
根据当前的信息判断，原因在于对 priviledgeMode 和 mstatus.mprv，以及例外代理的理解。按照测试程序的想法，当 dmode 为 S，那么例外被触发后，就应该被代理到 S 态处理。但是当前 CPU 的状态仍然为 M 态，是不应该被代理到 S 态的。

heiyuen1999, 2023-02-02:
> 您可以贴一下 CSR 模块内部的对应信号么，包括先不限于 priviledgeMode, m/s -cause, m/s status, m/s tval, csrio.exception.bits.uop.cf.exceptionVec, csrio.exception.valid.
> 
> 另外，您展示的测试程序是程序执行的全部么？我猜测是使用的 riscv-tests 的代码框架，但是似乎掏空了初始化流程。 riscv-tests 的初始化流程中，有对 CPU 特权级等状态的修改，这会影响到例外的执行。
以下是完整波形
[image]

heiyuen1999, 2023-02-02:
> 根据当前的信息判断，原因在于对 priviledgeMode 和 mstatus.mprv，以及例外代理的理解。按照测试程序的想法，当 dmode 为 S，那么例外被触发后，就应该被代理到 S 态处理。但是当前 CPU 的状态仍然为 M 态，是不应该被代理到 S 态的。
> 
> 特权级手册中提到：”Traps never transition from a more-privileged mode to a less-privileged mode.“，说明了这一场景的处理方式。

谢谢，确认了一下应该是我测试用例写的有问题。

closed 2023-02-02 by Lemover