OpenXiangShan/XiangShan #1915: 请问关于LDQ中提到的，虚地址前递失败的情况

Heresyrac, 2023-02-13:
文档中有提到说在很罕见的情况下会发现在数据前递时虚地址和实地址会匹配到不同的项，此时认为存在前递失败，因此需要进行重定向。
但是到底是怎样的一种罕见情况下会出现这种前递失败呢，文档里好像没有说明？当虚实地址的映射关系改变的时候难道不会清空store buffer/store Queue吗？能否具体说明一下？

AugustusWillisWang, 2023-02-14:
当两个虚地址同时映射到同一个实地址时，如果对这两个虚地址的写同时存在于 sq/sbuffer，前递就会出现这样的问题。

> 当虚实地址的映射关系改变的时候难道不会清空store buffer/store Queue吗？能否具体说明一下？

在用 store 指令更新页表时，硬件不会自动刷新 store buffer/store Queue。只有 SFENCE，satp 寄存器写等操作会触发 store buffer/store Queue 刷新。

Heresyrac, 2023-02-15:
> 都有可能 ：）

了解了。那么可以认为除此以外，没有其他的可能导致这种不一致的情况了吗？

Heresyrac, 2023-02-15:
> > > 都有可能 ：）
> > 
> > 
> > 了解了。那么可以认为除此以外，没有其他的可能导致这种不一致的情况了吗？
> 
> 是的。

明白了，感谢解答

Heresyrac, 2023-02-17:
> > > 都有可能 ：）
> > 
> > 
> > 了解了。那么可以认为除此以外，没有其他的可能导致这种不一致的情况了吗？
> 
> 是的。

关于这个我还想请问一个问题，有没有可能由于store修改了页表，导致之后流水线或者队列中的同时存在的两条指令，出现虚地址相同实地址不同的情况？
比如说有指令a store PTE b: store 1->x, c: store->x d: load x->，然后bcd三条都还没有提交. b取物理地址的时候读到了tlb中旧的页表内容，而在c和b中间的某个时刻，x的地址代换被tlb换出了，然后c执行的时候，ptw读到了新的页表内容，导致bc的物理地址不同？
另外d检查前递的时候，能否只检查实地址？

closed 2025-11-05 by Yan-Muzi