OpenXiangShan/XiangShan #1940: 关于浮点动态舍入模式的几个疑问

HUA-FENG1995, 2023-02-24:
[image]
①怎么打开动态舍入模式？是在指令后面加111吗，但这样好像编译不过。
②上面规范里关于“如果将frm设为无效值（101-111），则后续使用任何动态舍入模式执行浮点操作都将引发非法指令异常”怎么理解？是比如，先将frm设为（101-111），然后执行fadd.s fd, fs1,fs2就会报非法指令异常。但rm不为111，算是使用动态舍入模式的指令吗

ljwljwljwljw, 2023-02-24:
1. 请参考riscv汇编手册 https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md#floating-point-rounding-modes
2. 如果使用非动态舍入模式(指令中指定了有效的舍入模式)，则不是非法指令；如果指令使用动态舍入模式(指令中未指定舍入模式，使用frm的值)，且frm的值为101-111，则是非法指令

HUA-FENG1995, 2023-02-24:
> 1. 请参考riscv汇编手册 https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md#floating-point-rounding-modes
> 2. 如果使用非动态舍入模式(指令中指定了有效的舍入模式)，则不是非法指令；如果指令使用动态舍入模式(指令中未指定舍入模式，使用frm的值)，且frm的值为101-111，则是非法指令

也就是说如果指令中rm有指定的舍入模式，属于静态舍入模式（请问这时候受frm影响吗？），如果指令中的rm为rm没有规定舍入模式，这时候rm值默认为111，为动态舍入模式，使用frm的值，特别的当frm值为101-111时，此指令会引发非法指令异常。对吗？

ljwljwljwljw, 2023-02-26:
对的

closed 2023-03-11 by ljwljwljwljw