OpenXiangShan/XiangShan #1992: Merge CIRCT to nanhu

LandonWong, 2023-03-23:
您好，我在为nanhu分支增加circt时，编译emu(SimTop.v)时firtool报错如下

```
------------------------------------------------------------------------------
Error: firtool failed.
ExitCode:
1
STDOUT:
STDERR:
<stdin>:1:1: error: Unhandled annotation: {class = "freechips.rocketchip.util.ParamsAnnotation", params = {baseAddress = 939524096 : i64, intStages = 0 : i64}, paramsClassName = "freechips.rocketchip.devices.tilelink.CLINTParams", target = "SimTop.CLINT"}
circuit SimTop :
^
<stdin>:1:1: error: Unhandled annotation: {class = "freechips.rocketchip.util.RegFieldDescMappingAnnotation", regMappingSer = {baseAddress = 939524096 : i64, deviceName = "CLINT", 

...

<stdin>:1:1: error: Annotation targets non-wireable operation: {class = "firrtl.passes.wiring.SourceAnnotation", pin = "DISPLAY_LOG_ENABLE", target = "SimTop.SimTop.logEnable"}
circuit SimTop :
^
<stdin>:1:1: error: Unable to apply annotation: {class = "firrtl.passes.wiring.SourceAnnotation", pin = "DISPLAY_LOG_ENABLE", target = "SimTop.SimTop.logEnable"}
circuit SimTop :
^
<stdin>:1:1: error: Unable to resolve source for pin: "DISPLAY_LOG_ENABLE"
circuit SimTop :
^
------------------------------------------------------------------------------
```
移植过程按照 #1517 和 #1824 进行（除去bump chisel回3.5.0的修改，目前我的build.sc中chisel版本也为3.5.0）。我所使用的circtChisel版本也为0.6.0。在google查阅circtChisel相关问题后未果，尝试使用chisel-3.6.0-RC2很多语法并不兼容（circtChisel貌似已经合入Chisel）。

请问您是否熟悉以上报错，或问题可能出现在哪里？谢谢！

wakafa1, 2023-04-01:
目前香山对 CIRCT 的支持还不完善，一些 transform 需要在 CIRCT 中重新实现一遍；我们后续会跟进 CIRCT 的支持，敬请期待

eastonman, 2023-04-01:
我近日也尝试了在 master 分支使用 MFC 进行编译，目前遇到的问题与上述 nanhu 分支基本一致。
主要有三处

第一处为依赖项 rocketchip 中的 annotation CIRCT 不认识，体现为 firtool complains
```
<stdin>:1:1: error: Unhandled annotation: {class = "freechips.rocketchip.util.ParamsAnnotation", params = {baseAddress = 1006632960 : i64, intStages = 0 : i64, maxHarts = 15872 : i6
    4, maxPriorities = 7 : i64}, paramsClassName = "freechips.rocketchip.devices.tilelink.PLICParams", target = "SimTop.TLPLIC"}
```
经询问 @ZenithalHourlyRate Rocket 的编译选项中有"--disable-annotation-unknown"，此问题可以解决。

https://github.com/chipsalliance/rocket-chip/blob/3da8af49dd93a2d153379348edcb0c614559c151/build.sc#L123

第二处为涉及 Top.io 的 Boring，例如
```
    val logEnable = (timer >= io.logCtrl.log_begin) && (timer < io.logCtrl.log_end)
    ExcitingUtils.addSource(logEnable, "DISPLAY_LOG_ENABLE")
```
出错为
```
<stdin>:1:1: error: Annotation targets non-wireable operation: {class = "firrtl.passes.wiring.SourceAnnotation", pin = "DISPLAY_LOG_ENABLE", target = "SimTop.SimTop.logEnable"}
```

暂时没有想到 easy fix

第三处为 `XSLog` `XSAcummulate` 等 Debug Sugar 中大量使用的 multi-sink boring。
firtool complains not supported，如

```
SimTop.scala:100:24: error: This source is involved with a Wiring Problem where the source or the sink are multiply instantiated and this is not supported.
PerfCounterUtils.scala:44:31: note: The sink is here.
```
也暂时没有 easy fix

可参考的不涉及 Boring 将顶层信号传入所有 Module 的办法如 Rocket 的 BundleBridge

https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/diplomacy/BundleBridge.scala

eastonman, 2023-04-01:
以上 二、三 点问题可以通过关闭 perfDebug 暂时规避，但仍遇到 firtool assertion failed 问题
已向 CIRCT 提 issue：llvm/circt#4915

poemonsense, 2023-07-31:
This issue will be used to track our progress on updating the Chisel/Scala/Mill versions. The overall target is to support Chisel 5.0 and Scala 2.13 on XiangShan and the submodules, with CIRCT as the [only] Chisel compiler.

The proposed steps include:
1. Bump mill to the latest version
2. Bump Chisel to 3.5.6 for all modules
3. Bump Scala to 2.13 for all modules
4. Bump Chisel to 3.6.0 for all modules
5. Re-writing all the FIRRTL transforms in CIRCT
6. Bump Chisel to 5.0 for all modules (including the transforms)

After the last step, we will 100% migrate our build flows to CIRCT.

Estimated time to complete: December 31, 2023

@sequencer would you please help us review the steps? Thanks.

sequencer, 2023-07-31:
> Re-writing all the FIRRTL transforms in CIRCT

What transforms are xiangshan requiring for now?

poemonsense, 2023-07-31:
> > Re-writing all the FIRRTL transforms in CIRCT
> 
> What transforms are xiangshan requiring for now?

Only simple transforms in https://github.com/OpenXiangShan/XiangShan/tree/master/src/main/scala/xstransforms. IMHO we can even remove them directly.

sequencer, 2023-07-31:
BTW, try to use `Werror` for Xiangshan for the migration flow.

sequencer, 2023-08-02:
See chipsalliance/playground master branch. 
Chisel5 is already fully supported.

poemonsense, 2023-10-11:
We have just upgraded the build scripts (via PR #2372) to support Chisel 6.0.0-M3 with the CIRCT build tool. We will continue to optimizing the build flow in the next steps.

closed 2023-10-11 by poemonsense