OpenXiangShan/XiangShan #2011: Dispatch2Rs 的 LessExu 实现BUG

dingiso, 2023-03-31:
**Intoduction**

Dispatch2Rs 的 LessExu 中负载均衡的实现好像有问题，我理解的是输出端口应该可以根据 pinpong 的 false 和 true 自由的选择不同的输入端口进行连接，

但现在 pairIndex 的实现和使用方式

```scala
 val pairIndex = (0 until outer.numOut).map(i => (i + 2) % outer.numOut)
io.out(i) <> io.in(pairIndex(i))
```

导致选择的输入端口一定小于 numOut 也就是大于 numOut 的输入端口永远都不会被选择，这不符合负载均衡的理由

**Expected behavior**

输出端口在不同周期根据 pingpong 的值选择不同的输入端口进行连接

**Fix Code**

为了实现负载均衡我写了一段代码可以进行修复，尽量做到了编译期的优化。如果可以的话，后面可以提交 PR

```scala
// 上面的代码是未负载均衡的端口定义，当需要负载均衡时，下面的实现会进行覆盖

// 当输入端口大于输出端口时打开负载均衡
val enableLoadBalance = outer.numIn > outer.numOut
if(enableLoadBalance){
        // pingpong 只覆盖需要切换的端口，前面的端口不设置pinpong
        // 这样可以使 pingpong 的长度最小
	val numPingPongBits = outer.numIn - outer.numOut
	val pingpong = Seq.fill(numPingPongBits)(RegInit(false.B))
	pingpong.foreach(p=>p := !p)
        // 将少的输出对应到更多输入端口
	val pairIndex = (0 until outer.numOut).map(i =>  i + numPingPongBits)
	def needLoadBalance(index: Int): Bool = {
                //  需要进行切换的最小值
		val minIndex = 2 * outer.numOut - outer.numIn
                // 对应到 pingpong 的位置
		val bitIndex = index - minIndex
                // 只对需要切换的进行切换，否则返回false
		if(index >= minIndex) pingpong(bitIndex) else false.B
	}
        for ((config, i) <- outer.configs.zipWithIndex) {
          // 根据pinpong 动态切换
          when (needLoadBalance(i)) {
            // 切换
            io.out(i) <> io.in(pairIndex(i))
            if (config.length < numMaxExuConfig) {
              val thisCanAccept = config.map(_.canAccept(io.in(pairIndex(i)).bits.ctrl.fuType)).reduce(_ || _)
              io.out(i).valid := io.in(pairIndex(i)).valid && thisCanAccept
              io.in(pairIndex(i)).ready := io.out(i).ready && thisCanAccept
            }
            println(s"Dispatch2Rs ports balance between $i and ${pairIndex(i)}")
          }
        }
}
```

**例如**

1. 有 5 个输入[0,1,2,3,4]，3个输出[a,b,c]。 
2. 其中 (b,c) 可以在 (1,2) 和 (3,4) 中切换。pinpong 一共就是(5-3)一共两位。
3. pairIndex 的对应关系就是 (a,b,c) 对应 (2,3,4)， 但只会使用到后两位。 
4. minIndex 就是 1 也就是 b ，代表 (a,0) 对应关系永远不变，会优化一部分，从b开始切换。
5. bitIndex 将（b,c) 对应到 pingpong[0] 和 pingpong[1] 
6. 当需要切换的时候，会在 when 处动态的进行切换

Lemover, 2023-04-04:
感谢您能注意到这个细节。
Dispatch2RS 控制 Dispatch 到 RS 的输入-输出连接图。您提到，如果 numIn 如果大于 numOut，就会出现负载不均衡的错误。这个问题不会实际存在，因为 numIn 和 numOut 是相等的。
这里的负载均衡，不是指多数个 In 在少数个 Out 间均衡，而是指 In（dispatch） 到 Out（RS）的多个”通道“（port）间的均衡。因为一个 dispatch 可以负责多个 RS，每个 RS 对应一个或多个 port，而发射的多个 port 中，下标小的 port 更繁忙，下标大的 port 更空闲。port 的不平衡导致多个 RS 的利用也会出现不均衡，进而导致 执行单元的不均衡，因此需要在 dispatch2RS 阶段进行 负载均衡的优化。

dingiso, 2023-04-04:
Hi Lemover 

感谢您的解释，port 均衡是正确的，不过是否 pairIndex 使用 +2 作为负载均衡的设置过于粗糙，能否在这里做出优化呢

Lemover, 2023-04-04:
简单的 pairIndex+2，确实有点粗糙了。不过因为目前 port 宽度为 4，所以即使固定设置成 2，也可以基本满足均衡的要求了。
更激进的均衡算法，可能需要对 port 的利用率进行分析，如果 RS 的种类不同，也会增加这里的复杂度。如果再考虑到时序，和RS的一些出队入队的优化，会是一个比较复杂的问题。
这里确实可以是一个比较重要的优化点，感谢。

closed 2023-04-04 by dingiso