OpenXiangShan/XiangShan #2069: 请教一下 lr sc之间进行store是否合理？

kl123s, 2023-05-05:
**Is your feature request related to a problem? Please describe.**
香山南湖dcache对应地址块在接受到lr指令后，对该地址进行store操作，再进行sc指令操作会匹配成功，不会返回错误。 
疑问：1 单核的lr指令过后，store指令只在storebuffer上操作 ，无法立即写入 dcache，此原子操作功能是否有问题？
2 即使store指令能在lr锁定时间进入dcache，但目前没有看到内部代码对这进入的store指令检测的功能？
3 对于双核的原子指令操作：两个内核之间没有对这原子操作进行通信的功能，这样双核的原子操作功能需要怎样进行实现？

AugustusWillisWang, 2023-05-08:
1.2. RISC-V 手册没有规定同一 hart 内 program order 在 LR 之后的 store 对 SC 指令执行结果的影响。
3. 目前香山的原子操作靠某个核的数据缓存独占缓存行来实现。uncache 原子操作在南湖中尚不被支持。

kl123s, 2023-05-08:
> 1.2. RISC-V 手册没有规定同一 hart 内 program order 在 LR 之后的 store 对 SC 指令执行结果的影响。 3. 目前香山的原子操作靠某个核的数据缓存独占缓存行来实现。uncache 原子操作在南湖中尚不被支持。
好的谢谢。
针对第三个问题，cache地址空间的原子操作是在dcache内部进行实现，那如果多核的原子操作，核1对该地址块进行原子操作，该原子操作信息通过什么方式进行共享至其他核，来实现其多核的原子操作？
还是说，在核1进行原子操作的时候，其他核不会执行原子操作？

closed 2025-09-05 by Yan-Muzi