OpenXiangShan/XiangShan #2073: Questions about generating checkpoints

Changxi-Liu, 2023-05-08:
Hi, I am trying to do something based on NEMU's checkpoints method. 

When I try to generate the checkpoint for following steps:
1. ./build/riscv64-nemu-interpreter   -D ${HOME}/data/spec_cpt -w workloadName -C profiling  -b --simpoint-profile --cpt-interval 10 -u  -r ./resource/gcpt_restore/build/gcpt.bin  ./ready-to-run/linux-0xa0000.bin
2.  export CLUSTER=/home/lcx/data/spec_cpt/cluster/workloadName
3. ./resource/simpoint/simpoint_repo/bin/simpoint     -loadFVFile ${HOME}/data/spec_cpt/profiling/workloadName/simpoint_bbv.gz     -saveSimpoints $CLUSTER/simpoints0 -saveSimpointWeights $CLUSTER/weights0     -inputVectorsGzipped -maxK 30 -numInitSeeds 2 -iters 1000 -seedkm 123456 -seedproj 654321
4. ./build/riscv64-nemu-interpreter ./ready-to-run/linux-0xa0000.bin  -D ${HOME}/data/spec_cpt/ -w workloadName -C take_cpt -b ${HOME}/data/spec_cpt/cluster --cpt-interval 10

I am using the tracing branch and git id is 812c20b5088fcbb437ba07f773c11696fc8a1904.

At the 4th step, I got an error: riscv64-nemu-interpreter: src/monitor/monitor.c:298: init_monitor: Assertion `restorer != NULL' failed.

Do you have any idea how this happened?
Thanks for your information.

shinezyy, 2023-05-09:
The process of SimPoint checkpointing includes 3 individual steps

SimPoint Profiling to get BBVs. (To save space, they often output in compressed formats such as bbv.gz.)
SimPoint clustering. You can also opt to Python and sk-learn to do k-means clustering. (In this step, what is typically obtained are the positions selected by SimPoint and their weights.)
Taking checkpoints according to clustering results. (In the RVGCpt process, this step generates the checkpoints that will be used for simulation.)
If you have problem generating SPECCPU checkpoints, following links might help you.

[The video to build SPECCPU, put it in Linux, and run it in NEMU to get SimPoint BBVs](https://drive.google.com/file/d/1msr_YijlYN4rxpn71bod1LAoRWs5VtAL/view?usp=sharing) (step 1)
[The document to do SimPoint clustering based on BBVs and take simpoint checkpoints](https://zhuanlan.zhihu.com/p/604396330) (step 2 & 3)

**In you case, you should add "-r resource/gcpt_restore/build/gcpt.bin"**
**It seems that you are using --cpt-interval=10, which is too small.** When you are debugging this workflow, I suggest to use --cpt-interval=1000. And for performance estimation, It is typically 10M-100M.

Changxi-Liu, 2023-05-11:
@shinezyy  Thanks for your suggestions. now I can generate checkpoints.

abebeos, 2023-09-29:
@Changxi-Liu / @shinezyy , looks that this issue is resolved and can be closed

closed 2023-09-30 by shinezyy