OpenXiangShan/XiangShan #2114: 拿到 XSTop.v 之後如何在 FPGA 上面進行設計與驗證。

Chiwawachiwawa, 2023-05-31:
你好，請問如果我獲得了 XSTop.v 文件之後，我下一步該如何在 FPGA 上面進行驗證呢。
我目前是開了一個 vivado 的專案，並且新增了一個 block designs，目前的思路為添加了XSTop.v之後將一些外設補齊，但是我發現我沒有辦法將 XSTop.v 加入到我的設計之中，想請問我該如何解決，亦或是我的方向與步驟做錯了，需要做額外的準備，感謝你!

wakafa1, 2023-06-01:
XSTop.v 作为香山处理器核的 RTL 代码，还需要搭建一版 SoC 才能在 FPGA 上进行验证；SoC 需要支持的内容包括但不限于 DDR 控制器、中断连线、各类外设比如串口等

Chiwawachiwawa, 2023-06-01:
感謝回復!那請問香山生成的東西，我只需要將 XSTop.v 加入vivado 專案之中，其他外設自行定義，還是需要添加其他生成的資源呢?

wakafa1, 2023-06-01:
还有一些 array 开头的 verilog 文件也需要包含在内；
另外需要注意，生成出来的 verilog 可能有部分不可综合的代码，请手动把 DebugOptions 中的 AlwaysBasicDiff 参数设置为 false

Chiwawachiwawa, 2023-06-01:
了解了，感謝你的細心回復!有遇到問題還要再麻煩了!

Chiwawachiwawa, 2023-06-01:
> 请手动把 DebugOptions 中的 AlwaysBasicDiff 参数设置为 false

不好意思，請問是 vivado 之中的選項嗎(? 我在2022.2的版本中沒有找到

Tang-Haojin, 2023-06-01:
是香山的参数，请参见[这一行代码](https://github.com/OpenXiangShan/XiangShan/blob/2c1a69a06196b2ec5ef3d8623ab178b077ad3281/src/main/scala/xiangshan/Parameters.scala#L296)。

Chiwawachiwawa, 2023-06-04:
你好，修改完後依舊有錯誤的檔案，大多為sim相關的

wakafa1, 2023-06-05:
请提供更加详细的信息以便我们判断