OpenXiangShan/XiangShan #2123: make verilog error

Singleboardcomputer, 2023-06-08:
**Describe the bug**
make verilog -j8
mkdir -p build
time -a -o ./build/time.log mill -i XiangShan.runMain top.TopMain -td build  \
        --config DefaultConfig                                        \
        --infer-rw --repl-seq-mem -c:top.TopMain:-o:build/XSTop.v.conf --gen-mem-verilog full
                         \
        --num-cores 1                                  \
        --disable-all --remove-assert --fpga-platform
Cannot resolve $file import: /root/riscv_cpu/XiangShan/rocket-chip/api-config-chipsalliance/build-rules/mill/build.sc
make: *** [Makefile:79: build/XSTop.v] Error 1

**To Reproduce**
https://github.com/OpenXiangShan/XiangShan/issues/931

**Expected behavior**
get the generated.v

**Environment (optional, if necessary):**
 WSL2 - Ubuntu 20.04.6 LTS

wakafa1, 2023-06-09:
Have you executed `make init` command successfully before building?

Singleboardcomputer, 2023-06-14:
Thanks a lot , the issue is solved

closed 2023-06-14 by Singleboardcomputer